xArm-Developer/xarm_ros2 #1: Bug when running xarm_moveit_config demo for simulated robotic arm

syahmi001, 2021-08-24:
Hello and Hi.

I encountered some errors when running the xarm_moveit_config demo.
First, somehow I can't move the robotic arm or even change its orientation. Usually in move-it package, you should have some kind of arrows at the end of the TCP. Here I include the picture as ref:

[image]

The robotic is just there. I can't change its orientation or create new orientation. When I press the "Plan" button in RVIZ, nothing happens. But when I press the "Plan and Execute", the RVIZ windows immediately closed. Here is the log when that happens:
(Note: This is for the Simulated Arm case)

```
[move_group-4] You can start planning now!
[move_group-4]
[move_group-4] [INFO] [1629786605.311458700] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-4] [INFO] [1629786605.311723100] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[rviz2-2] [INFO] [1629786605.311863900] [move_group_interface]: Plan and Execute request accepted
[rviz2-2] terminate called after throwing an instance of 'std::runtime_error'
[rviz2-2]   what():  Node has already been added to an executor.
[ERROR] [rviz2-2]: process has died [pid 508, exit code -6, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /home/syahmi/dev_ws/install/xarm_moveit_config/share/xarm_moveit_config/rviz/moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_9ghplhlt --params-file /tmp/launch_params_8arke03w -r /tf:=tf -r /tf_static:=tf_static'].
[move_group-4] [INFO] [1629786605.325969600] [moveit_move_group_default_capabilities.move_action_capability]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[move_group-4] [INFO] [1629786605.326192900] [moveit_move_group_default_capabilities.move_action_capability]: Goal constraints are already satisfied. No need to plan or execute any motions
[move_group-4] [INFO] [1629786605.326243400] [moveit_move_group_default_capabilities.move_action_capability]: Requested path and goal constraints are already met.
```

My environment for the test is Ubuntu 20.04 + ROS Foxy with all the requirement installed as per the guide.
Did I miss something?

penglongxiang, 2021-08-24:
Hi @syahmi001 , I see some warnings (yellow caption) of Motion planning add-on in your picture. Have you properly installed Moveit2?

vimior, 2021-08-24:
Which command are you running? Can you provide detailed output of the following startup process and operations

syahmi001, 2021-08-24:
Hi @penglongxiang. Yes I think I did install Moveit2. But I install it via binary installation, not by source. I don't think there is any different between installing the package either by source or by binary installation.

The code line that I used to run the demo is below:

```
$ cd ~/dev_ws/
$ source install/setup.bash
$ ros2 launch xarm_moveit_config xarm6_moveit_fake.launch.py [add_gripper:=true]
```

vimior, 2021-08-24:
Can you provide me with the command you started to execute? And the detailed output of the startup process and operation process?

vimior, 2021-08-24:
@syahmi001 I installed moveit2 through the binary installation package, and this problem has been reproduced. We need some time to find specific problems and fixes. Please be patient and wait for the update or try to install moveit2 from the source code. Thank you for your feedback.

syahmi001, 2021-08-24:
Yeah thanks. Really appreciate it!

vimior, 2021-08-24:
Hi, @syahmi001 We found through debugging that joint_trajectory_controller/JointTrajectoryController is not installed, you can try again after installing ros2-controllers.
`sudo apt install ros-foxy-ros2-controllers`

syahmi001, 2021-08-24:
Yeah, that did solve the issue. I guess you need to manually install the plugins in ROS2, different from ROS1. Thanks!

closed 2021-08-24 by syahmi001