xArm-Developer/xarm_ros2 #3: How to call motion planning service with a smooth pose path array?

scottnortman, 2021-09-01:
I have the simulated arm working with the xarm_planner and moveit.  I have written a python node to call the services PlanJoint and PlanPose, but I need the ability to execute a smooth cartesian motion with an array of poses I generate in a different node.

I see the c++ methods

planPoseTargets and planCartesianPath

but it does not seem those methods are exposed via service calls.

Is there a way to pass an array of Poses via a service call?

vimior, 2021-09-02:
Hi @scottnortman 
Several service calls (xarm_exec_plan, xarm_pose_plan, xarm_joint_plan) are provided in xarm_plan_node, refer to xarm_planner/test/test_xarm_planner_client_joint.cpp.
Or instantiate xarm_planner directly in the code, so that you can directly use the methods provided by xarm_planner (executePath, planPoseTarget, planJointTarget), refer to xarm_planner/test/test_xarm_planner_api_joint.cpp.

scottnortman, 2021-09-03:
I added the service to call the cartesian planner; it works as I need for  my application

See my fork here:  https://github.com/scottnortman/xarm_ros2

closed 2021-09-03 by scottnortman