xArm-Developer/xarm_ros2 #6: Cannot execute planned moves "Unable to identify any set of controllers that can actuate the specified joints".

GuiMateus, 2022-01-06:
I've been trying to get this driver to run in my machine, however it seems that I am running into some problems with executing planned moves.  I understand that the repo is targeted for Foxy, but I am attempting to run it in Galactic. For reference, the UR driver for ROS2 plans and executes moves correctly, so does the Xarm driver for ROS-Noetic.

In order to get this repo to run for galactic I had modify xarm_controller/include/xar_controller/fake_xarn_hw.h and xarm_controller/include/xar_controller/xarm_hw.h since some of the functions from hardware_interface don't seem to be virtual in Galactic, causing the building process to fail if not modified.

Screenshots:
Path planning is successful and controller is not found:
[image]
This also caught my attention:
[image]

System:

- ROS2 Galactic
- Pop!OS 20.04

Full logs:

> [INFO] [launch]: All log files can be found below /home/guilherme/.ros/log/2022-01-06-12-25-18-151700-pop-os-84212
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [robot_state_publisher-1]: process started with pid [84225]
> [INFO] [rviz2-2]: process started with pid [84227]
> [INFO] [static_transform_publisher-3]: process started with pid [84229]
> [INFO] [move_group-4]: process started with pid [84231]
> [INFO] [joint_state_publisher-5]: process started with pid [84233]
> [INFO] [ros2_control_node-6]: process started with pid [84235]
> [INFO] [spawner.py-7]: process started with pid [84237]
> [static_transform_publisher-3] [INFO] [1641468319.330395189] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'link_base'
> [robot_state_publisher-1] Link link_base had 1 children
> [robot_state_publisher-1] Link link1 had 1 children
> [robot_state_publisher-1] Link link2 had 1 children
> [robot_state_publisher-1] Link link3 had 1 children
> [robot_state_publisher-1] Link link4 had 1 children
> [robot_state_publisher-1] Link link5 had 1 children
> [robot_state_publisher-1] Link link6 had 1 children
> [robot_state_publisher-1] Link link_eef had 0 children
> [robot_state_publisher-1] [INFO] [1641468319.333149556] [robot_state_publisher]: got segment link1
> [robot_state_publisher-1] [INFO] [1641468319.333217231] [robot_state_publisher]: got segment link2
> [robot_state_publisher-1] [INFO] [1641468319.333229663] [robot_state_publisher]: got segment link3
> [robot_state_publisher-1] [INFO] [1641468319.333238812] [robot_state_publisher]: got segment link4
> [robot_state_publisher-1] [INFO] [1641468319.333248240] [robot_state_publisher]: got segment link5
> [robot_state_publisher-1] [INFO] [1641468319.333259205] [robot_state_publisher]: got segment link6
> [robot_state_publisher-1] [INFO] [1641468319.333269821] [robot_state_publisher]: got segment link_base
> [robot_state_publisher-1] [INFO] [1641468319.333282043] [robot_state_publisher]: got segment link_eef
> [robot_state_publisher-1] [INFO] [1641468319.333291611] [robot_state_publisher]: got segment world
> [ros2_control_node-6] terminate called after throwing an instance of 'std::runtime_error'
> [ros2_control_node-6]   what():  Wrong state or command interface configuration.
> [ros2_control_node-6] missing state interfaces:
> [ros2_control_node-6] ' joint1/position '	' joint1/velocity '	' joint2/position '	' joint2/velocity '	' joint3/position '	' joint3/velocity '	' joint4/position '	' joint4/velocity '	' joint5/position '	' joint5/velocity '	' joint6/position '	' joint6/velocity '	
> [ros2_control_node-6] missing command interfaces:
> [ros2_control_node-6] ' joint1/position '	' joint1/velocity '	' joint2/position '	' joint2/velocity '	' joint3/position '	' joint3/velocity '	' joint4/position '	' joint4/velocity '	' joint5/position '	' joint5/velocity '	' joint6/position '	' joint6/velocity '	
> [move_group-4] [WARN] [1641468319.360900506] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
> [move_group-4] [WARN] [1641468319.360960569] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
> [move_group-4] [INFO] [1641468319.363261741] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00215583 seconds
> [move_group-4] [INFO] [1641468319.363292750] [moveit_robot_model.robot_model]: Loading robot model 'xarm6'...
> [move_group-4] [INFO] [1641468319.363304623] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
> [move_group-4] Link link_base had 1 children
> [move_group-4] Link link1 had 1 children
> [move_group-4] Link link2 had 1 children
> [move_group-4] Link link3 had 1 children
> [move_group-4] Link link4 had 1 children
> [move_group-4] Link link5 had 1 children
> [move_group-4] Link link6 had 1 children
> [move_group-4] Link link_eef had 0 children
> [spawner.py-7] [INFO] [1641468319.493458806] [spawner_xarm6_traj_controller]: Waiting for /controller_manager services
> [ERROR] [ros2_control_node-6]: process has died [pid 84235, exit code -6, cmd '/home/guilherme/rosWorkspaces/urdriver/install/controller_manager/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_owlpg5uc --params-file /home/guilherme/rosWorkspaces/xarmdriver_no_update/install/xarm_controller/share/xarm_controller/config/xarm6_controllers.yaml'].
> [move_group-4] [INFO] [1641468319.559645313] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
> [move_group-4] [INFO] [1641468319.559888148] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
> [move_group-4] [INFO] [1641468319.560327445] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
> [move_group-4] [INFO] [1641468319.560989462] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
> [move_group-4] [INFO] [1641468319.561042471] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[... 151 lines cut ...]
> [move_group-4] [INFO] [1641468947.693487768] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRT.cpp:227 - xarm6/xarm6: Created 20 states
> [move_group-4] [INFO] [1641468947.693704553] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRT.cpp:227 - xarm6/xarm6: Created 32 states
> [move_group-4] [INFO] [1641468947.694120383] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRT.cpp:227 - xarm6/xarm6: Created 38 states
> [move_group-4] [INFO] [1641468947.694205449] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/tools/multiplan/src/ParallelPlan.cpp:135 - ParallelPlan::solve(): Solution found by one or more threads in 0.002622 seconds
> [move_group-4] [INFO] [1641468947.694406310] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRT.cpp:119 - xarm6/xarm6: Starting planning with 1 states already in datastructure
> [move_group-4] [INFO] [1641468947.694446957] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRT.cpp:119 - xarm6/xarm6: Starting planning with 1 states already in datastructure
> [move_group-4] [INFO] [1641468947.695154650] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRT.cpp:227 - xarm6/xarm6: Created 11 states
> [move_group-4] [INFO] [1641468947.697715490] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRT.cpp:227 - xarm6/xarm6: Created 44 states
> [move_group-4] [INFO] [1641468947.697895119] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/tools/multiplan/src/ParallelPlan.cpp:135 - ParallelPlan::solve(): Solution found by one or more threads in 0.003590 seconds
> [move_group-4] [INFO] [1641468947.697998693] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/src/SimpleSetup.cpp:179 - SimpleSetup: Path simplification took 0.000012 seconds and changed from 2 to 2 states
> [move_group-4] [INFO] [1641468947.699620877] [moveit_move_group_default_capabilities.move_action_capability]: Motion plan was computed succesfully.
> [rviz2-2] [INFO] [1641468948.014958580] [move_group_interface]: Planning request accepted
> [rviz2-2] [INFO] [1641468948.115675622] [move_group_interface]: Planning request complete!
[... cut ...]
> [move_group-4] [ERROR] [1641468949.663023468] [moveit_ros.trajectory_execution_manager]: Unable to identify any set of controllers that can actuate the specified joints: [ joint1 joint2 joint3 joint4 joint5 joint6 ]
> [move_group-4] [ERROR] [1641468949.663051125] [moveit_ros.trajectory_execution_manager]: Known controllers and their joints:
> [move_group-4] 
> [rviz2-2] [INFO] [1641468950.063991212] [move_group_interface]: Execute request accepted
> [rviz2-2] [INFO] [1641468950.164584847] [move_group_interface]: Execute request aborted
> [rviz2-2] [ERROR] [1641468950.264831095] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
>

MinnaZhong, 2022-01-07:
Hi GuiMateus,

Sorry, Galatic is not compatible yet, will let you know once it is supported.

vimior, 2022-01-17:
Hi @GuiMateus 

Now it is preliminarily compatible with the ROS2 Galactic version. Please switch to the galactic branch. If you have any questions during the use process, please contact us. Thank you for your support.

GuiMateus, 2022-01-17:
> Hi @GuiMateus
> 
> Now it is preliminarily compatible with the ROS2 Galactic version. Please switch to the galactic branch. If you have any questions during the use process, please contact us. Thank you for your support.

Works perfectly now, thanks :)

closed 2022-01-17 by GuiMateus