xArm-Developer/xarm_ros2 #9: error: \u2018hardware_interface::return_type xarm_control::FakeXArmHW::configure(const hardware_interface::HardwareInfo&)\u2019 marked \u2018override\u2019, but does not override

alienshek337, 2022-04-07:
[image]

got this issue while colcon build

vimior, 2022-04-11:
@alienshek337 
Can you provide the ros version you are currently using and which branch of xarm_ros2 are you using?

alienshek337, 2022-04-11:
i am using ros foxy and I followed the instruction from the read so it should be master branch

vimior, 2022-04-11:
@alienshek337 
Ok, I updated ros foxy to the latest test

vimior, 2022-04-11:
@alienshek337 
Hello, I have upgraded ros foxy here and I have not reproduced this problem for the time being. Can you check the version of the ros2_control package of your computer?
`apt search ros-foxy-ros2-control && apt search ros-foxy-hardware-interface`

alienshek337, 2022-04-12:
[image]

vimior, 2022-04-12:
@alienshek337 
Hello, I am not sure whether the version of ros-foxy-hardware-interface does not correspond to the cause, you can try to upgrade ros-foxy-hardware-interface
`sudo apt upgrade  ros-foxy-hardware-interface ros-foxy-ros2-control-test-asserts`

alienshek337, 2022-04-12:
I upgraded the ros-foxy-hardware-interface and I got new error:
[image]