xArm-Developer/xarm_ros2 #12: ROS2 Humble colcon build error

ivanrulik, 2022-09-12:
Hi I hope I can get some help to fix this. Currently I'm trying to test this library but I get the following error:

--- stderr: xarm_controller                                  
In file included from /home/ivan/dev_ws/src/xarm_ros2/xarm_controller/src/hardware/uf_robot_system_hardware.cpp:9:
/home/ivan/dev_ws/src/xarm_ros2/xarm_controller/include/xarm_controller/hardware/uf_robot_system_hardware.h:21:10: fatal error: hardware_interface/types/hardware_interface_status_values.hpp: No such file or directory
   21 | #include "hardware_interface/types/hardware_interface_status_values.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/ivan/dev_ws/src/xarm_ros2/xarm_controller/src/hardware/uf_robot_fake_system_hardware.cpp:9:
/home/ivan/dev_ws/src/xarm_ros2/xarm_controller/include/xarm_controller/hardware/uf_robot_fake_system_hardware.h:21:10: fatal error: hardware_interface/types/hardware_interface_status_values.hpp: No such file or directory
   21 | #include "hardware_interface/types/hardware_interface_status_values.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Any help or advice is greatly apreciated

vimior, 2022-09-13:
Hi, @ivanrulik 
Please use the code of the corresponding branch, [Humble](https://github.com/xArm-Developer/xarm_ros2/tree/humble)

ivanrulik, 2022-09-13:
Thank you very much, everything works fine now.

closed 2022-09-13 by ivanrulik