xArm-Developer/xarm_ros2 #13: Error launching xarm_moveit_servo to simulate xArm with keyboard input ROS2 Humble

ivanrulik, 2022-09-14:
Hi! sorry to bother again so quick.
I have been trying to run: 
- ros2 launch xarm_moveit_servo xarm_moveit_servo_fake.launch.py
- ros2 run xarm_moveit_servo xarm_keyboard_input

But the simulated xArm has not moved. Here are captures of RViz and rqt_graph if it helps to show my problem more clearly:
[image]

[image]

Any help or advice is greatly appreciated

penglongxiang, 2022-09-19:
Hi @ivanrulik, it seems MoveitServo consider this initial position as singularity point. Maybe you have to move the second last joint a little bit first to make the IK solver able to find a solution. Please have a try.

cemoke, 2024-09-09:
Hi, I also have the same problem, and moving the second last joint doesn't work either. In the terminal where I ran the first launch file, I get the following error:

[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'servo_server' of type 'moveit_servo::ServoServer' in container '/xarm_moveit_servo_container': Failed to find class with the reques
ted plugin name. 

Maybe that is causing the issue. Please help as this is a needed feature on my side

penglongxiang, 2024-09-10:
> Hi, I also have the same problem, and moving the second last joint doesn't work either. In the terminal where I ran the first launch file, I get the following error:
>
> [ERROR] [...]

@cemoke If you see this error message, **maybe you did not have moveit-servo package installed**. Please check the availability by 
```shell
$ dpkg -l | grep moveit-servo
```
If nothing shows up. You could install it by
```shell
$ sudo apt-get install ros-$ROS_DISTRO-moveit-servo
```

penglongxiang, 2024-09-10:
@HansRen1024 Please make `xarm_keyboard_input` node shell active and move the target joint by the number keys (1-7). Use '`R`' key for reverse direction.

penglongxiang, 2024-09-12:
@cemoke Sorry I could not replicate your issue and it worked OK on my side. Could you confirm that you have no problem following the tutorial examples [here](https://moveit.picknik.ai/humble/doc/examples/realtime_servo/realtime_servo_tutorial.html)?

penglongxiang, 2024-09-14:
Hi @cemoke, it turns out my local code version was not updated. We will be working on this issue and get back to you once solved.

penglongxiang, 2024-09-18:
@cemoke, we have fixed the issue for humble branch, please update the repo and see if it works, thank you.

penglongxiang, 2024-09-20:
Hi @cemoke, could you provide more information for this error? Like starting and target joint poses, the way robot is mounted and the payload at end-effector. In fact a video showing the robot status and motion would be much preferred. You can send it to support@ufactory.cc and quote this issue page, we will evaluate ASAP.

penglongxiang, 2024-09-21:
@cemoke please also attach your log package downloadable from Ufactory Studio (Settings -> My Device -> Log), it will contain additional info we need. Thanks.

cemoke, 2024-10-04:
I sent the email, and I have attached a video showing my configuration. As a reference for others, I would like to share my observations and questions.

First of all, why is the home configuration near the robot's singularity? This does not make any sense to me. Is there an easy way to change it or specify it as an argument to the launch file so that it does not result in an immediate singular configuration, as it is currently doing?

Secondly, I am using the default configuration without any modifications: the robot is mounted horizontally (as it should be), and there is nothing attached to the robot. However, the example you provided does not work, and the robot is giving me an error stating that the speed limit is exceeded:

[ros2_control_node-3] [WARN] [1728055196.648868521] [UFACTORY.RobotHW]: [192.168.1.164] set_servo_angle_j, ret= 1
[ros2_control_node-3] [WARN] [1728055196.654975660] [UFACTORY.RobotHW]: [192.168.1.164] Robot State detected! State: 5
[ros2_control_node-3] [ERROR] [1728055196.687712088] [UFACTORY.RobotHW]: [192.168.1.164] UFACTORY Error detected! Code C24 -> [ Speed Exceeds Limit ]

This does not make sense to me, as it suggests that the default configuration you provide results in a violation of the speed limit. Is there a way I can increase the speed limit so that this error does not occur?

Since I am not changing anything and am simply running the example provided in the documentation, I believe that if you run the same example on your setup, you should encounter the same issue. I used the following launch file:

ros2 launch xarm_moveit_servo lite6_moveit_servo_realmove.launch.py robot_ip:=192.168.1.123 joystick_type:=3

I tested both with the space mouse and the keyboard input, as well as for both joint and task space commands, and still received the same error.

Please help me resolve this issue as soon as possible. I would greatly appreciate a prompt response.

penglongxiang, 2024-10-07:
Hi, @cemoke, we still need your **log package** as well to finally diagnose the issue. After downloading the log, please try upgrading your controller firmware to the latest version (**V2.5.5**). It might be due to an older firmware that the servo mode has not been optimized, and I have tested your operations in the video with latest firmware with no problem.

As for the singularity of zero-joint home position, we have designed a proper "initial position" for Lite 6 in UFACTORY studio that you can always return to (as a default configuration), you can also define a different default joint position in your program for your application to avoid initial singularity.

cemoke, 2024-10-07:
> Hi, @cemoke, we still need your **log package** as well to finally diagnose the issue. After downloading the log, please try upgrading your controller firmware to the latest version (**V2.5.5**). It [...]

Thank you for the prompt response.

You can access the logs here: https://drive.google.com/file/d/18Su3hj5oOHgW0SurnrWFiSpncp8ftP07/view?usp=sharing

I'll update the firmware and let you know whether the issue has resolved. 

Thanks

cemoke, 2024-10-07:
It appears that the original issue was resolved by following these steps:

    Updating the firmware to version V2.5.5.
    Starting from the initial configuration defined in UFACTORY Studio.
    Reducing the speed to 30%.

Thank you very much for your prompt responses and all your help.

However, occasionally, if I trigger another error—such as self-collision or using the emergency stop—I cannot continue using the teleoperation node after clearing the error in UFACTORY Studio and releasing the emergency stop. To proceed, I need to restart both the robot and the node. Is there a way to clear the error and continue operation using ROS?

penglongxiang, 2024-10-08:
@cemoke In your log I noticed you have enabled the reduce mode, and limited the linear speed to be 50 mm/s. This may trigger the over-speed error since moveit servo is not aware of this additional limitation in the controller box. You may turn off the reduce mode or change the joint and linear speed limit to a proper value.

As for the method to continue servoing after error reported, we have not yet found a proper way to update the actual robot state to the ROS2 controller, if just setting the robot mode back to 1 and state back to 0, the robot may suddenly jump back to the position before the error occurs, which is not safe compared with restarting the application. Maybe there is a way to reset in ROS2, but I have no experience with it by far.