xArm-Developer/xarm_ros2 #15: Failed to colcon build xarm_controller

Jinzhou511, 2022-09-27:
**When I tried to colcon build xarm in step 4.5, I encountered following issue, it seems that I don't have the hpp file named "hardware_interface/types/hardware_interface_status_values.hpp", which led to fail to build xarm_controller.** 

**Terminal information:**

walt@WaltUbuntu:~/dev_ws$ colcon build --parallel-workers 2
_Starting >>> xarm_controller
stderr: xarm_controller                              
In file included from /home/walt/dev_ws/src/xarm_ros2/xarm_controller/src/hardware/uf_robot_fake_system_hardware.cpp:9:
/home/walt/dev_ws/src/xarm_ros2/xarm_controller/include/xarm_controller/hardware/uf_robot_fake_system_hardware.h:21:10: fatal error: hardware_interface/types/hardware_interface_status_values.hpp: No such file or directory
   21 | #include "hardware_interface/types/hardware_interface_status_values.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/walt/dev_ws/src/xarm_ros2/xarm_controller/src/hardware/uf_robot_system_hardware.cpp:9:
/home/walt/dev_ws/src/xarm_ros2/xarm_controller/include/xarm_controller/hardware/uf_robot_system_hardware.h:21:10: fatal error: hardware_interface/types/hardware_interface_status_values.hpp: No such file or directory
   21 | #include "hardware_interface/types/hardware_interface_status_values.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/uf_robot_hardware.dir/build.make:76: CMakeFiles/uf_robot_hardware.dir/src/hardware/uf_robot_system_hardware.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/uf_robot_hardware.dir/build.make:90: CMakeFiles/uf_robot_hardware.dir/src/hardware/uf_robot_fake_system_hardware.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/uf_robot_hardware.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

Failed   <<< xarm_controller [0.79s, exited with code 2]_

Jinzhou511, 2022-09-27:
I'm using Ubuntu 22 and Ros2 Humble

vimior, 2022-09-28:
Hi, @Jinzhou511 
ros2 humble please switch to [humble](https://github.com/xArm-Developer/xarm_ros2/tree/humble) branch

closed 2022-10-20 by vimior