xArm-Developer/xarm_ros2 #16: xarm_gazebo not working atm

dinkoosmankovic, 2022-10-15:
Is it me or xarm gazebo simulation is still not working? When I run the launch file, it says that gzserver dies. Debug does not reveal any new info.

vimior, 2022-10-21:
@dinkoosmankovic 
Hi, can you provide the ros version you are using and which is the start command and the log of the problem?

dinkoosmankovic, 2022-10-24:
Hi,

I'm using ROS2 Humble inside an Ubuntu 22.04 GUI capable contatiner. My other gazebo simulations work fine. This one started working after update, but not as intended. By running **ros2 launch xarm_moveit_config xarm6_moveit_gazebo.launch.py** I get following:

- Robot transforms is not ok, robot is lying on the floor and not on the table with initial position
- Log is:
 
`[ERROR] [rviz2-5]: process has died [pid 6260, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /root/etf-xarm-lab/install/xarm_moveit_config/share/xarm_moveit_config/rviz/moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_ubsit4dz --params-file /tmp/launch_params_6uw0etam --params-file /tmp/launch_params_aonav0a_ -r /tf:=tf -r /tf_static:=tf_static'].
[INFO] [move_group-7]: sending signal 'SIGINT' to process[move_group-7]
[INFO] [static_transform_publisher-6]: sending signal 'SIGINT' to process[static_transform_publisher-6]
[INFO] [robot_state_publisher-3]: sending signal 'SIGINT' to process[robot_state_publisher-3]
[INFO] [gzclient-2]: sending signal 'SIGINT' to process[gzclient-2]
[INFO] [gzserver-1]: sending signal 'SIGINT' to process[gzserver-1]
[move_group-7] [INFO] [1666604357.254005312] [rclcpp]: signal_handler(signum=2)
[static_transform_publisher-6] [INFO] [1666604357.259612804] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-3] [INFO] [1666604357.268945833] [rclcpp]: signal_handler(signum=2)
[move_group-7] [INFO] [1666604357.298911003] [moveit.ros_planning_interface.moveit_cpp]: Deleting MoveItCpp
[move_group-7] [INFO] [1666604357.299947391] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopped publishing maintained planning scene.
[move_group-7] [INFO] [1666604357.301188991] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping world geometry monitor
[move_group-7] [INFO] [1666604357.301719318] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
[move_group-7] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[move_group-7]          at line 127 in ./src/class_loader.cpp
[move_group-7] Stack trace (most recent call last):
[move_group-7] #16   Object "", at 0xffffffffffffffff, in 
[move_group-7] #15   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x55bf70898794, in 
[move_group-7] #14   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3cbf8fee3f, in __libc_start_main
[move_group-7] #13   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3cbf8fed8f, in 
[move_group-7] #12   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x55bf708976cb, in 
[move_group-7] #11   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x55bf70899779, in 
[move_group-7] #10   Object "/opt/ros/humble/lib/libmoveit_cpp.so.2.5.3", at 0x7f3cc02259c6, in moveit_cpp::MoveItCpp::~MoveItCpp()
[move_group-7] #9    Object "/opt/ros/humble/lib/libmoveit_cpp.so.2.5.3", at 0x7f3cc0223929, in 
[move_group-7] #8    Object "/opt/ros/humble/lib/libmoveit_trajectory_execution_manager.so.2.5.3", at 0x7f3cbf5db535, in trajectory_execution_manager::TrajectoryExecutionManager::~TrajectoryExecutionManager()
[move_group-7] #7    Object "/opt/ros/humble/lib/libmoveit_trajectory_execution_manager.so.2.5.3", at 0x7f3cbf5ee2f9, in 
[move_group-7] #6    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f3cbfea15fc, in rclcpp::Node::~Node()
[move_group-7] #5    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f3cbfea15ce, in rclcpp::Node::~Node()
[move_group-7] #4    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f3cbfe7c079, in 
[move_group-7] #3    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f3cbfea2b29, in 
[move_group-7] #2    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f3cbfea2a70, in rclcpp::node_interfaces::NodeBase::~NodeBase()
[move_group-7] #1    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f3cbfe7c079, in 
[move_group-7] #0    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f3cbfea1aa1, in 
[move_group-7] Segmentation fault (Address not mapped to object [0x7f3c8e2107a8])
[INFO] [static_transform_publisher-6]: process has finished cleanly [pid 6262]
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 6256]
[INFO] [gzserver-1]: process has finished cleanly [pid 6252]
[INFO] [launch]: process[gzserver-1] was required: shutting down launched system
[ERROR] [move_group-7]: process has died [pid 6264, exit code -11, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_epexdgo2 --params-file /tmp/launch_params_9oulvcsx --params-file /tmp/launch_params_jbcwib22 --params-file /tmp/launch_params_08w310rc --params-file /tmp/launch_params_069xx0m5 --params-file /tmp/launch_params_jlze5v2j --params-file /tmp/launch_params_zudd0csi'].
[move_group-7] 
[ERROR] [gzclient-2]: process[gzclient-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [gzclient-2]: sending signal 'SIGTERM' to process[gzclient-2]
[ERROR] [gzclient-2]: process has died [pid 6254, exit code -15, cmd 'gzclient'].
[INFO] [launch]: process[gzclient-2] was required: shutting down launched system`

vimior, 2022-10-25:
@dinkoosmankovic 
Before you run, check whether there are any remaining gazebo processes
`ps -aux | grep gz`
If yes, please kill the remaining processes before running

vimior, 2022-10-28:
@dinkoosmankovic 
Please pull the latest code, then delete the compilation directory and try to recompile

dinkoosmankovic, 2022-10-29:
Did so, now I get this error:

`[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 24116]
[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'OnProcessExit' object has no attribute '_OnActionEventBase__on_event'
[INFO] [robot_state_publisher-3]: sending signal 'SIGINT' to process[robot_state_publisher-3]
[INFO] [gzclient-2]: sending signal 'SIGINT' to process[gzclient-2]
[INFO] [gzserver-1]: sending signal 'SIGINT' to process[gzserver-1]
[robot_state_publisher-3] [INFO] [1667051329.940801760] [rclcpp]: signal_handler(signum=2)
[gzserver-1] [INFO] [1667051329.949872371] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
[ERROR] [gzclient-2]: process has died [pid 24112, exit code -2, cmd 'gzclient'].
[INFO] [launch]: process[gzclient-2] was required: shutting down launched system
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 24114]
[INFO] [gzserver-1]: process has finished cleanly [pid 24110]
[INFO] [launch]: process[gzserver-1] was required: shutting down launched system`

However, if I comment out the following in _robot_beside_table_gazebo.launch.py:

`RegisterEventHandler(
            event_handler=OnProcessExit(
                target_action=gazebo_spawn_entity_node,
                on_exit=load_controllers,
            )
        ),`

the simulation starts but the robot has no normal TFs (visualized as broken in gazebo).

dinkoosmankovic, 2022-10-29:
This seems to be an issue with xarm6. Lite6 and xarm7 are working fine.

dinkoosmankovic, 2022-10-30:
Update: it's a bug in gazebo, after upgrading it works well, although sometimes it does not start well. We can close this now.

closed 2022-10-30 by dinkoosmankovic