xArm-Developer/xarm_ros2 #19: move real xarm7 dual arm with moveit not work

k-ish0212, 2022-11-29:
Hi,

I run a dual xarm7 launch file as below but it output an error and moveit doesn't work.

ros2 launch xarm_moveit_config dual_xarm7_moveit_realmove.launch.py robot_ip_1:=192.168.10.220 robot_ip_2:=192.168.10.211 add_gripper_1:=true add_gripper_2:=true

The logs are below.

If anyone knows how to solve this, please tell me,
Thank you in advance.

`[INFO] [launch]: All log files can be found below /root/.ros/log/2022-11-29-17-42-04-158774-xarm-dual-ros2-14545
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [14553]
[INFO] [rviz2-2]: process started with pid [14555]
[INFO] [static_transform_publisher-3]: process started with pid [14557]
[INFO] [move_group-4]: process started with pid [14559]
[INFO] [joint_state_publisher-5]: process started with pid [14561]
[INFO] [ros2_control_node-6]: process started with pid [14563]
[INFO] [spawner.py-7]: process started with pid [14565]
[INFO] [spawner.py-8]: process started with pid [14567]
[static_transform_publisher-3] [INFO] [1669743725.571346982] [static_transform_publisher]: Spinning until killed publishing transform from 'ground' to 'link_base'
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link L_link_base had 1 children
[robot_state_publisher-1] Link L_link1 had 1 children
[robot_state_publisher-1] Link L_link2 had 1 children
[robot_state_publisher-1] Link L_link3 had 1 children
[robot_state_publisher-1] Link L_link4 had 1 children
[robot_state_publisher-1] Link L_link5 had 1 children
[robot_state_publisher-1] Link L_link6 had 1 children
[robot_state_publisher-1] Link L_link7 had 1 children
[robot_state_publisher-1] Link L_link_eef had 1 children
[robot_state_publisher-1] Link L_xarm_gripper_base_link had 5 children
[robot_state_publisher-1] Link L_left_outer_knuckle had 1 children
[robot_state_publisher-1] Link L_left_finger had 0 children
[robot_state_publisher-1] Link L_link_tcp had 0 children
[robot_state_publisher-1] Link L_left_inner_knuckle had 0 children
[robot_state_publisher-1] Link L_right_inner_knuckle had 0 children
[robot_state_publisher-1] Link L_right_outer_knuckle had 1 children
[robot_state_publisher-1] Link L_right_finger had 0 children
[robot_state_publisher-1] Link R_link_base had 1 children
[robot_state_publisher-1] Link R_link1 had 1 children
[robot_state_publisher-1] Link R_link2 had 1 children
[robot_state_publisher-1] Link R_link3 had 1 children
[robot_state_publisher-1] Link R_link4 had 1 children
[robot_state_publisher-1] Link R_link5 had 1 children
[robot_state_publisher-1] Link R_link6 had 1 children
[robot_state_publisher-1] Link R_link7 had 1 children
[robot_state_publisher-1] Link R_link_eef had 1 children
[robot_state_publisher-1] Link R_xarm_gripper_base_link had 5 children
[robot_state_publisher-1] Link R_left_outer_knuckle had 1 children
[robot_state_publisher-1] Link R_left_finger had 0 children
[robot_state_publisher-1] Link R_link_tcp had 0 children
[robot_state_publisher-1] Link R_left_inner_knuckle had 0 children
[robot_state_publisher-1] Link R_right_inner_knuckle had 0 children
[robot_state_publisher-1] Link R_right_outer_knuckle had 1 children
[robot_state_publisher-1] Link R_right_finger had 0 children
[robot_state_publisher-1] [INFO] [1669743725.577911461] [robot_state_publisher]: got segment L_left_finger
[robot_state_publisher-1] [INFO] [1669743725.578072792] [robot_state_publisher]: got segment L_left_inner_knuckle
[robot_state_publisher-1] [INFO] [1669743725.578089845] [robot_state_publisher]: got segment L_left_outer_knuckle
[robot_state_publisher-1] [INFO] [1669743725.578097805] [robot_state_publisher]: got segment L_link1
[robot_state_publisher-1] [INFO] [1669743725.578105326] [robot_state_publisher]: got segment L_link2
[robot_state_publisher-1] [INFO] [1669743725.578112115] [robot_state_publisher]: got segment L_link3
[robot_state_publisher-1] [INFO] [1669743725.578118464] [robot_state_publisher]: got segment L_link4
[robot_state_publisher-1] [INFO] [1669743725.578125466] [robot_state_publisher]: got segment L_link5
[robot_state_publisher-1] [INFO] [1669743725.578131911] [robot_state_publisher]: got segment L_link6
[robot_state_publisher-1] [INFO] [1669743725.578138055] [robot_state_publisher]: got segment L_link7
[robot_state_publisher-1] [INFO] [1669743725.578144695] [robot_state_publisher]: got segment L_link_base
[robot_state_publisher-1] [INFO] [1669743725.578150968] [robot_state_publisher]: got segment L_link_eef
[robot_state_publisher-1] [INFO] [1669743725.578157334] [robot_state_publisher]: got segment L_link_tcp
[... 348 lines cut ...]
[move_group-4] *     - GetPlanningSceneService
[move_group-4] *     - KinematicsService
[move_group-4] *     - MoveAction
[move_group-4] *     - MotionPlanService
[move_group-4] *     - QueryPlannersService
[move_group-4] *     - StateValidationService
[move_group-4] ********************************************************
[move_group-4] 
[move_group-4] [INFO] [1669743786.887832844] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[move_group-4] [INFO] [1669743786.887851477] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context initialization complete
[move_group-4] Loading 'move_group/ApplyPlanningSceneService'...
[move_group-4] Loading 'move_group/ClearOctomapService'...
[move_group-4] Loading 'move_group/MoveGroupCartesianPathService'...
[move_group-4] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-4] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[move_group-4] Loading 'move_group/MoveGroupKinematicsService'...
[move_group-4] Loading 'move_group/MoveGroupMoveAction'...
[move_group-4] Loading 'move_group/MoveGroupPlanService'...
[move_group-4] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-4] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-4] 
[move_group-4] You can start planning now!
[move_group-4] 
`
[image]

vimior, 2022-11-30:
@k-ish0212 
Hi, please confirm that your robotic arm IP can be connected. You can first confirm whether the IPs of the two robotic arms can be connected.
```bash
ros2 launch xarm_api xarm7_driver.launch.py robot_ip:=192.168.10.220
ros2 launch xarm_api xarm7_driver.launch.py robot_ip:=192.168.10.211
```

k-ish0212, 2022-11-30:
oh, that was my mistake. 192.168.10.220 was wrong. 192.168.10.210 was precise.

it worked for dual arms, thank you!
[image]

closed 2022-12-01 by k-ish0212