xArm-Developer/xarm_ros2 #20: displacement in relative

tvchau, 2022-12-01:
Hi, 
I am quite new to ROS and i was trying to move my xarm6 arm with a displacement in relative, for example with a delta Z = 20mm.
I was trying the command
`ros2 service call /xarm/set_position xarm_msgs/srv/MoveCartesian "{pose: [0, 0, 20, 0, 0, 0], speed: 50, acc: 500, mvtime: 0, relative: true}"`
but it doesn't do the job, it was moving to its zero position instead of a simple displacement in Z.

Could you advice me on that ?

vimior, 2022-12-02:
@tvchau 
Since some msgs are shared by multiple services, some parameters are not supported. For details, please refer to the description of msg.
Previously, the set_position on ros did not support relative parameters. I just updated the support for some parameters. You can update the following code to recompile and use.
thank you for your support

vimior, 2022-12-02:
@tvchau 
Some parameters require a certain version of the firmware to support. If some parameters are found to be invalid or not as expected during use, it is recommended to update the firmware to the latest version.