xArm-Developer/xarm_ros2 #21: When launching

wxgisu, 2022-12-26:
I'm trying to follow the instruction to setup and run the ros2 humble xarm application. When executing `ros2 launch xarm_description xarm6_rviz_display.launch.py add_gripper:=true`, I've seen following error: 

```
[INFO] [launch]: All log files can be found below /home/xiaoguang/.ros/log/2022-12-26-12-41-34-380288-xiaoguang-desktop-1729766
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable '[<launch.substitutions.text_substitution.TextSubstitution object at 0x7f6d80d78e50>]' not found on the PATH
```

I have already run `source ~/dev_ws/install/setup.zsh` in my terminal to ensure env variables are properly set.

penglongxiang, 2023-01-03:
Hi @wxgisu, we did not reproduce this issue in humble environment. Have you checked if you have problem running other ROS2 packages or the official tutorials? If the problem persists, maybe you can try re-installing ROS2 environment.

wxgisu, 2023-01-25:
Hi @penglongxiang, thanks for your response. It turned out that I was using a built from source version of ros2, instead of the binary one installed by apt. Once I switched to the binary one and rebuilt the xarm workspace, things worked. Thanks!

closed 2023-01-25 by wxgisu