xArm-Developer/xarm_ros2 #22: How to assign IP to a simulator?

wxgisu, 2023-01-28:
Hello, I am trying to setup the xarm gazebo simulation and control it by making service calls with xarm_ros_client. However when I launch the xarm_api nodes with `ros2 launch xarm_api xarm6_driver.launch.py robot_ip:=192.168.1.117`, the TCP socket fails to bind. Then I realized it is trying to connect to the real robot with an IP address. Is there anyway I can assign an IP to the gazebo simulation, so I start visualize the API effects? Or is there anyway to do this without an IP?

vimior, 2023-01-30:
@wxgisu 
The gazebo simulation is used in the ros package we provide, which is not related to the real robotic arm, so there is no need to provide IP, but all services and APIs under xarm_api cannot be used. You can control it through moveit's move_group API

wxgisu, 2023-02-01:
Thanks for the answer @vimior. So if I want to create a digital twin of the xarm, and try to write some controlling software to complete some simple pick and place tasks, should I use the xarm_sdk instead of the xarm_api? 

Even if I have a real robot, I still need to have some simulation env to develop and test my code, right?

vimior, 2023-02-02:
@wxgisu 
If you are writing control software in ROS, it is recommended to use the services provided by xarm_api. 
xarm_sdk is a submodule pointing to the C++ SDK, which has nothing to do with ROS. 
xarm_api actually encapsulates the api of the sdk into a ros service. 
If not ROS applications can directly use the [C++ SDK](https://github.com/xArm-Developer/xArm-CPLUS-SDK).
Both xarm_api and xarm_sdk can only be used on real devices.
If you need to simulate in ROS, you cannot directly use the xarm_api service or the SDK API, you can only use the moveit API, and the moveit API is effective in both simulation and real environments.

wxgisu, 2023-02-20:
Thanks for the clarification. Closing this ticket.

closed 2023-02-20 by wxgisu