xArm-Developer/xarm_ros2 #24: How to achieve the maximum speed of xarm using Moveit Servoing

Akumar201, 2023-02-02:
Hi, I am using moveit xarm tutorial to move the arm using joystick, I would like to know how can I achieve the maximum speed from the xarm using move-it serving.

vimior, 2023-02-03:
@Akumar201 
Here is the use of servoj mode to control the robotic arm. It does not support specifying the speed, but the speed can be controlled by sending frequency.
Here I provide an example of modifying some configuration parameters. You can refer to the modification and debugging to see what speed meets your needs.
1. Modify _*xarm_controller/config/xarm6_controllers.yaml*_ (note that the 6-axis is used as an example), and appropriately increase the `update_rate` (100?)
2. Modify _*xarm_moveit_servo/config/xarm_moveit_servo_config.yaml*_, increase publish_period appropriately (0.1?), set `low_latency_mode` to true

Akumar201, 2023-02-03:
Thank You ! It worked , however I would like to know that what would be the limits of these values ?

vimior, 2023-02-08:
@Akumar201 
`update_rate` is the frequency at which the instructions of the controller (such as JointTrajectoryController) are updated to the hardware interface (uf_robot_hardware).
`publish_period` is used internally by moveit_servo, mainly to scale the sent instructions and control the frequency sent to move_group (if `low_latency_mode` is true, it will not control the frequency sent)
For more details, you can check the description or code of moveit_servo