xArm-Developer/xarm_ros2 #25: Attempting to Save Recorded Trajectory Crashses API

Adam-Sawyer, 2023-02-05:
After recording a trajectory on my xArm6 whenever I attempt to save it the API immediately crashes

[image]
[image]

vimior, 2023-02-06:
@Adam-Sawyer 
The trajectoryis saved in the control box, the filename parameter only needs the file name, not the path.

Adam-Sawyer, 2023-02-06:
> WARNING: This message has originated from an External Source. This may be a phishing expedition that can result in unauthorized access to our IT System. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Is it possible for me to access the trajectory recording directly?

vimior, 2023-02-06:
@Adam-Sawyer 
For recording files, the SDK only has interfaces for saving and playback
If you need to read the file content, you can use xArmStudio