xArm-Developer/xarm_ros2 #27: Question: Hardware Interface Code

peterdavidfagan, 2023-02-10:
Currently the Hardware interfaces code is not compatible with MoveItConfigsBuilder as it incorrectly parses the robot_ip from the robot_description. The substr index when changed to 0 fixed this problem (see linked sections of codebase).

https://github.com/xArm-Developer/xarm_ros2/blob/67af13d70fda6de299de630cc540e3a2d271e489/xarm_controller/src/hardware/uf_robot_system_hardware.cpp#L65

vimior, 2023-02-13:
@peterdavidfagan 
Sorry, I have never used MoveItConfigsBuilder. If it is convenient, can you provide a short test code, mainly to test the correctness of robot_ip acquisition.
The parameters of the hardware interface layer are passed through URDF (such as the xarm6_ros2_control.xacro file), and the robot_ip parameter passed by the URDF to the hardware interface layer actually adds an R prefix on the basis of the actual robot_ip, so when parsing robot_ip at the hardware interface layer It should be substr(1), I'm still not sure how your parsing error happened.

peterdavidfagan, 2023-02-13:
Thanks for getting back to me on this @vimior.

> the robot_ip parameter passed by the URDF to the hardware interface layer actually adds an R prefix on the basis of the actual robot_ip, so when parsing robot_ip at the hardware interface layer It should be substr(1), I'm still not sure how your parsing error happened.

Ah ok I see, this is where the difference arises, MoveItConfigsBuilder directly parses from the URDF file but it doesn't expect there to be an added prefix.