xArm-Developer/xarm_ros2 #29: Gripper ROS Interface

peterdavidfagan, 2023-02-15:
Are there plans to support interfacing with the vacuum gripper via ROS and MoveIt. I might be missing something but I didn't find code for this in the existing repository.

peterdavidfagan, 2023-02-16:
Hi @vimior,

More than happy to support the use of the lite6 especially with MoveIt. 

Is there any support for interacting with the vacuum gripper even through the python SDK? I have seen that the studio app provides support.

vimior, 2023-02-17:
@peterdavidfagan 
It can be controlled through the API of the SDK or the service of ROS, but it cannot be controlled through the API of moveit for the time being.

peterdavidfagan, 2023-02-17:
Thanks @vimior, I appreciate your quick responses to these questions.

Elmessiry, 2024-11-20:
> @peterdavidfagan It can be controlled through the API of the SDK or the service of ROS, but it cannot be controlled through the API of moveit for the time being.

Hi, does this mean i cannot use the ros2 api for the vacuum control for the lite6?
if you dont mind pointing me to how I can control the vacuum gripper in a python script, for some reason i couldn't manage to do it with the ros2 services either

chipmunk321, 2025-04-25:
@penglongxiang is there any update or time plan regarding the vacuum gripper support in simulation (gazebo)?