xArm-Developer/xarm_ros2 #30: No planning library loaded error in RViz

efevreowner, 2023-02-17:
When I launch moveit for controlling robot in rviz using
`ros2 launch xarm_moveit_config lite6_moveit_realmove.launch.py robot_ip:=192.168.1.161 [add_gripper:=false]`
I get the following terminal output:
`ros2 launch xarm_moveit_config lite6_moveit_realmove.launch.py robot_ip:=192.168.1.161 [add_gripper:=false]
[INFO] [launch]: All log files can be found below /home/efevre/.ros/log/2023-02-17-09-51-41-333584-efevre-HP-250-G7-Notebook-PC-3972
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [3979]
[INFO] [rviz2-2]: process started with pid [3981]
[INFO] [static_transform_publisher-3]: process started with pid [3983]
[INFO] [move_group-4]: process started with pid [3985]
[INFO] [joint_state_publisher-5]: process started with pid [3987]
[INFO] [ros2_control_node-6]: process started with pid [3989]
[INFO] [spawner-7]: process started with pid [3991]
[static_transform_publisher-3] [WARN] [1676620302.468997493] []: Old-style arguments are deprecated; see --help for new-style arguments
[rviz2-2] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[static_transform_publisher-3] [INFO] [1676620302.544327965] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-3] from 'world' to 'link_base'
[robot_state_publisher-1] [INFO] [1676620302.553618567] [robot_state_publisher]: got segment link1
[robot_state_publisher-1] [INFO] [1676620302.553778913] [robot_state_publisher]: got segment link2
[robot_state_publisher-1] [INFO] [1676620302.553797982] [robot_state_publisher]: got segment link3
[robot_state_publisher-1] [INFO] [1676620302.553808071] [robot_state_publisher]: got segment link4
[robot_state_publisher-1] [INFO] [1676620302.553820098] [robot_state_publisher]: got segment link5
[robot_state_publisher-1] [INFO] [1676620302.553833061] [robot_state_publisher]: got segment link6
[robot_state_publisher-1] [INFO] [1676620302.553843178] [robot_state_publisher]: got segment link_base
[robot_state_publisher-1] [INFO] [1676620302.553856308] [robot_state_publisher]: got segment link_eef
[robot_state_publisher-1] [INFO] [1676620302.553867185] [robot_state_publisher]: got segment world
[ros2_control_node-6] [INFO] [1676620302.570274477] [resource_manager]: Loading hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[move_group-4] [WARN] [1676620302.624756436] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-4] Error:   Semantic description is not specified for the same robot as the URDF
[move_group-4]          at line 664 in ./src/model.cpp
[move_group-4] [INFO] [1676620302.628187047] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00313896 seconds
[move_group-4] [INFO] [1676620302.628649556] [moveit_robot_model.robot_model]: Loading robot model 'xarm6'...
[move_group-4] [INFO] [1676620302.628664038] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ros2_control_node-6] [INFO] [1676620302.659572163] [resource_manager]: Initialize hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[ros2_control_node-6] [INFO] [1676620302.727819626] [UFACTORY.RobotHW]: [192.168.1.161] namespace: /
[ros2_control_node-6] [INFO] [1676620302.727853514] [UFACTORY.RobotHW]: [192.168.1.161] robot_type: lite, hw_ns: ufactory, prefix: , report_type: normal
[ros2_control_node-6] [INFO] [1676620302.729262722] [UFACTORY.RobotHW]: [192.168.1.161] dof: 6, velocity_control: 0, add_gripper: 0, baud_checkset: 1, default_gripper_baud: 2000000
[ros2_control_node-6] [INFO] [1676620302.730921240] [ufactory_driver]: robot_ip=192.168.1.161, report_type=normal, dof=6
[ros2_control_node-6] [INFO] [1676620302.731110565] [ufactory_driver]: baud_checkset: 1, default_gripper_baud: 2000000
[ros2_control_node-6] SDK_VERSION: 1.11.6
[ros2_control_node-6] Tcp control connection successful
[spawner-7] [INFO] [1676620302.914577647] [spawner_lite6_traj_controller]: Waiting for '/controller_manager' services to be available
[ros2_control_node-6] ROBOT_IP: 192.168.1.161, VERSION: v1.12.2, PROTOCOL: V1, DETAIL: 6,9,LI1006,DL1000,v1.12.2, TYPE1300: [0, 0]
[ros2_control_node-6] Tcp Report Rich connection successful
[ros2_control_node-6] change prot_flag to 3
[ros2_control_node-6] Tcp Report Norm connection successful
[ros2_control_node-6] report_data_size: 494, size_is_not_confirm: 0
[ros2_control_node-6] report_data_size: 145, size_is_not_confirm: 0
[ros2_control_node-6] [INFO] [1676620303.062561017] [ufactory_driver]: [TCP STATUS] CONTROL: 1, REPORT: 1
[rviz2-2] [INFO] [1676620303.068054836] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1676620303.068645093] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[ros2_control_node-6] [INFO] [1676620303.101035136] [ufactory_driver]: gripper_speed: 2000, gripper_max_pos: 850, gripper_frequency : 10, gripper_threshold: 3, gripper_threshold_times: 10
[ros2_control_node-6] [INFO] [1676620303.124792322] [UFACTORY.RobotHW]: [192.168.1.161] System Sucessfully configured!
[ros2_control_node-6] [INFO] [1676620303.124829199] [resource_manager]: Successful initialization of hardware 'uf_robot_hardware/UFRobotSystemHardware'
[ros2_control_node-6] [INFO] [1676620303.125125389] [resource_manager]: 'configure' hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[ros2_control_node-6] [INFO] [1676620303.125139420] [resource_manager]: Successful 'configure' of hardware 'uf_robot_hardware/UFRobotSystemHardware'
[ros2_control_node-6] [INFO] [1676620303.125150464] [resource_manager]: 'activate' hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[rviz2-2] [INFO] [1676620303.134509068] [rviz2]: Stereo is NOT SUPPORTED
[move_group-4] [INFO] [1676620303.136165548] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-4] [INFO] [1676620303.136332435] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-4] [INFO] [1676620303.137756226] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-4] [INFO] [1676620303.138648686] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-4] [INFO] [1676620303.138677274] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-4] [INFO] [1676620303.139396483] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-4] [INFO] [1676620303.139421121] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-4] [INFO] [1676620303.140215470] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-4] [INFO] [1676620303.140946265] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-4] [WARN] [1676620303.141359543] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[... 92 lines cut ...]
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[rviz2-2] Warning: Invalid frame ID "link5" passed to canTransform argument source_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[rviz2-2] Error:   Semantic description is not specified for the same robot as the URDF
[rviz2-2]          at line 664 in ./src/model.cpp
[rviz2-2] [INFO] [1676620306.646173725] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00135559 seconds
[rviz2-2] [INFO] [1676620306.646231900] [moveit_robot_model.robot_model]: Loading robot model 'xarm6'...
[rviz2-2] [INFO] [1676620306.646243154] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-2] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[... cut ...]
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[rviz2-2] [ERROR] [1676620306.746748009] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.joint1.max_velocity' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.joint1.max_velocity} is of type {double}, setting it to {string} is not allowed.`

In RViz it shows error NO PLANNING LIBRARY LOADED and planning group is empty.

wxgisu, 2023-02-20:
It looks like there is a data type error: 
```
{robot_description_planning.joint_limits.joint1.max_velocity} is of type {double}, setting it to {string} is not allowed.
```

vimior, 2023-02-22:
@efevreowner 
What version of ROS are you using?

efevreowner, 2023-02-22:
@vimior ROS2 humble

vimior, 2023-02-22:
@efevreowner 
Maybe it's a problem with encoding settings, you can refer to some issues of moveit2 to see if it can be solved.
https://github.com/ros-planning/moveit2/issues/1596
https://github.com/ros-planning/moveit2/issues/1782

DaniGarciaLopez, 2023-02-24:
I'm also experiencing the same issue in Humble. I tried changing to cyclonedds and UTF8 encoding with no luck.