xArm-Developer/xarm_ros2 #31: MoveIt2 Rolling Integration

emfebert, 2023-03-24:
Hi,

Is there already a moveit2 rolling integration?

Thanks a lot!

vimior, 2023-03-27:
@emfebert 
There is no integration for the time being, you can try to use the code of the [humble](https://github.com/xArm-Developer/xarm_ros2/tree/humble) branch directly, and we will support `rolling` later when we have time.

peterdavidfagan, 2023-10-16:
Hey @vimior any updates on this?

penglongxiang, 2023-10-17:
Please checkout our provided ["rolling" branch](https://github.com/xArm-Developer/xarm_ros2/tree/rolling). However, it is recommended to use a stable ROS version like humble, since rolling is a developing version and will be constantly updated, we may not get noticed and catch up with its change.

peterdavidfagan, 2023-10-17:
Thanks for this confirmation @penglongxiang.