xArm-Developer/xarm_ros2 #33: Not able to change the speed of joints of xArm6 with Moveit

hmiranda-queiros, 2023-05-08:
Hello,

I am controlling an xArm6 with MoveIt on the `humble` branch by using your `xarm_planner` package. However even after increasing the default values of  the file `xarm_moveit_config/xarm6/joint_limits.yaml` or by increasing the `maxV_scale_factor` in the `xarm_planner/src/xarm_planner.cpp` file, I was not able to increase the speed or the acceleration of the real robot. I have this problem for all the types of path planners, i.e Carthesian, Pose and Joint targets.
Could you please tell what values I need to change in order to increase the acceleration and the speed of the xArm6 when it is controlled with MoveIt.
Thanks in advance for your help.

vimior, 2023-05-09:
@hmiranda-queiros 
Sorry, the `maxV_scale_factor` in xarm_planner is only set in planCartesianPath, so it doesn't take effect.
You can get the latest xarm_ros2 code, and then define `max_velocity_scaling_factor` and `max_acceleration_scaling_factor` in xarm_planner.cpp, you can adjust the speed and acceleration by modifying these two values.

hmiranda-queiros, 2023-05-18:
Hello again, I was able to change the speed of `planCartesianPath` by directly modifying the `joint_limit.yaml` of the xArm6 in the `xarm_moveit_config` package, as I read on the doc that sadly `max_velocity_scaling_factor` and `max_acceleration_scaling_factor` do not have effect on the `planCartesianPath` in MoveIt.
Thank you for your help.

closed 2023-05-18 by hmiranda-queiros