xArm-Developer/xarm_ros2 #34: C19 error when using tgpio service

KylecheungKml, 2023-05-19:
Hello. We are using ROS2 to develop our application on xArm6. The firmware version is 1.12.10, studio version is 1.12.4.

We added two new services in xarm driver_service to call API to control our end effector.

[image]
Then, we executed ros2 launch xarm_planner lite6_planner_realmove.launch.py robot_ip:=192.168.1.117 and executed a test script to demonstrate the workflow by calling services:[/xarm_pose_plan, /xarm_joint_plan, /xarm_exec_plan] (moving to a specific position and turning the end effector on/off).

[image]

However, after successfully executing the while loop several times, we encountered the C19 error code, causing the robot_state ROS topic to change (mode 1->0; state 2->5). Additionally, the joint_state topic stopped publishing. The xarm6_realmove log shows that the path was successfully planned, but could not be executed. But the end effector services keep working well. Here is the log:
[image]

We are sure that this case only occur involve calling tgoio service. Can you please assist us in resolving this issue? Thanks.

vimior, 2023-05-19:
@KylecheungKml 
Generally, C19 occurs at the end of the communication failure. There are many reasons for the failure. I guess you are timed out here (the default is 5ms). You can set the timeout time through set_tgpio_modbus_timeout.
Once an error occurs, the robotic arm cannot be controlled because the mode and status of the robotic arm have changed. The error must be cleared before the mode and status can be set.

KylecheungKml, 2023-05-22:
Thank you. The issue was resolved after increasing the tgpio timeout. However, after running the loop for two hours, the log showed the message: **_[ros2_control_node-6] [INFO] [1684733867.884816963] [UFACTORY.RobotHW]: [192.168.1.237] Read() returns: 5_**. As a result, the arm was unable to execute the planned pose. I also noticed that this issue arises when I boot up my computer for the first time and execute the _ros2 launch xarm_planner lite6_planner_realmove.launch.py robot_ip:=192.168.1.237_.  But the second-time execution is OK.

vimior, 2023-05-22:
@KylecheungKml 
1. Isn’t your robotic arm xArm6? Why do you use the lite6 startup script?
2. Can you check the SDK version output at startup?
3. Can't start xarm_planner for the first time to plan and output anything?

KylecheungKml, 2023-05-22:
@vimior 
1, sorry, it is a typo, should be xArm6.
2, SDK version is 1.11.7
3, Can't start xarm_planner problem does not always occur. Terminal keeps output [Read() returns: 5], Rviz2 will show up, but cannot exec the robot arm.
[image]

vimior, 2023-05-22:
@KylecheungKml 
I tried to use the same version to test, but I can't reproduce this situation for the time being. You can leave any abnormal output before and after the earliest output "Read() returns 5". Generally, this sentence is output because the correct response data cannot be obtained when obtaining the current position of the robotic arm.

KylecheungKml, 2023-05-22:
I have attached two log files, one for abnormal execution during the first run and another for normal execution during the second run. Based on the logs, I guess that starting with a C2 error at the beginning of the execution may cause the problem. However, it's unclear if the Read() returns 5 error that occurs during the two-hour execution also has this same issue.
[abnormal.txt](https://github.com/xArm-Developer/xarm_ros2/files/11528751/abnormal.txt)
[normal.txt](https://github.com/xArm-Developer/xarm_ros2/files/11528752/normal.txt)

vimior, 2023-05-22:
@KylecheungKml 
When C2 appears, the emergency stop is usually pressed or the emergency stop IO is triggered. The xarm_planner needs to be initialized when it is started. It cannot be initialized when it is in C2. The subsequent acquisition of the position of the robotic arm will also fail, but generally it will not return 5, I will try to reproduce the "Read() returns 5" phenomenon.

KylecheungKml, 2023-06-06:
Thank you. We have conducted tests for several weeks, and the issue with the "return 5" problem no longer occurs.

closed 2023-06-06 by KylecheungKml