xArm-Developer/xarm_ros2 #38: Object grasping problem

klima7, 2023-05-26:
I'm facing strange problem with grasping objects. On the video bellow You can see what is happening. I'm controlling arm with Moveit.

https://github.com/xArm-Developer/xarm_ros2/assets/52660101/b47db6e4-4f4a-449c-a684-8c004c389708

klima7, 2023-05-30:
I see that in [xarm_ros repo](https://github.com/xArm-Developer/xarm_ros) `gazebo_grasp_fix` plugin is used to grasp objects. Aren't there any alternatives for ROS 2?

penglongxiang, 2023-05-31:
It seems this plugin is not compatible with ROS2 yet, currently we have not found a good solution for this issue. Will keep an eye on it.

HansRen1024, 2024-10-15:
I got the same issue with this.

https://github.com/user-attachments/assets/35123a5b-da59-42f6-83e1-4558b692f3b2

penglongxiang, 2024-11-21:
@HansRen1024 [here](https://github.com/kongoncharuk/gazebo-pkgs/tree/humble) is a ROS2 ported version of `gazebo_grasp_fix` plugin. It will help attach the object to gripper when there are opposing forces detected. However, it seems not to be as stable as the behavior on ROS1 gazebo, the gripper still deforms when the fingers are in contact with the object while the attaching function works fine, not sure which part causes this issue. You may try giving a proper position command to the gripper according to the object size, thus making the deformation as small as possible.