xArm-Developer/xarm_ros2 #39: dual xarm realmove , humble , running both arms simultaneously

DailyNir, 2023-08-03:
i am using the dual xarm6 realmove launch commands and it is working great, i was wondering if it would be possible controlling both arms simultaneously because i couldn't figure it out .

please help, thanks.

vimior, 2023-08-07:
@DailyNir 
When you say controlling two arms at the same time, do you mean one trigger, and both arms move to the same position?
It is possible in theory, but since the planning of moveit depends on the position of the robotic arm, if the position feedback of two robotic arms is used, the effect may not be very good. You can try to use only one robotic arm A for planning, and the other robotic arm B for follow-up control , that is, send control commands to multiple robotic arms at the same time at the driver layer, or write another node to monitor the position of robotic arm A and control robotic arm B.
Note that this can only control multiple robotic arms at the same time, not full synchronization.

DailyNir, 2023-08-07:
I meant 2 separate path planners , sending commands separately but run simultaneously so they could preform tasks together .

vimior, 2023-08-09:
@DailyNir 
Yes, two path planners correspond to two robotic arms, but the control logic needs to be planned by itself, for example, using moveit's MoveGroupInterface control.

DailyNir, 2023-08-09:
yes, how would you go about it? because in the ros2 wrapper for the dual arms, you created movegroup for each arm separately but i cant use 2 motion planners simultaneously, please suggest a way

vimior, 2023-08-10:
@DailyNir 
Sorry, the control started by "dual_xxx" actually integrates the two robotic arms into a whole, so there is only one planner. Although different robotic arms correspond to different Planning Groups, only one can be used at the same time, which may not satisfy your scenario.

DailyNir, 2023-08-10:
so how would you create one movegroup for both arms to plan for both of them?

vimior, 2023-08-14:
Since a whole can only have one planner, if you need multiple planners, you can try to start multiple startup scripts that control a single robotic arm (`prefix:=L_ ros_namespace:=L`), and use MoveGroupInterface to connect to different MoveGroup to control different robotic arm.

DailyNir, 2023-08-14:
So I don't mind having one planner , but can it plan for both simultaneously? I mean plan and execute both with one planner?

vimior, 2023-08-14:
@DailyNir 
One planner can only plan and execute one at a time.

DailyNir, 2023-08-17:
ok, can you please point me to the services i can send to {x,y,z,r,p,y}  and than plan and execute within the rviz/gazebo dual xarm6? i am trying to create a code the plans and executed within the world of rviz but can find the relevant service to do so in your documentation . in rviz2 when i take the marker some x,y,z,r,p,y and i press the plan and execute button, i want to recreate it via code . let me know if this is possible .

vimior, 2023-08-24:
@DailyNir 
Sorry for taking so long to reply to you.
Refer to moveit's [MoveGroupInterface](https://moveit.picknik.ai/humble/doc/examples/move_group_interface/move_group_interface_tutorial.html), and correspondingly, please refer to our xarm_planner package (this package is only for the robotic arm, but the way of writing is similar)