xArm-Developer/xarm_ros2 #41: Finding the height of the work area.

SyllogismRXS, 2023-08-28:
Is there an example of finding the height of the work area relative to the robot's base_link by lowering the gripper until it touches the work area? The feedback would be provided by either the current sensing in the xarm's joints or through an added force torque sensor. If so, what interfaces in `xarm_msgs` should I use? Thanks!

penglongxiang, 2023-08-29:
Hi, there is currently no direct way of finding the touch point without a external Force/Torque (FT) sensor. The example code you mentioned is for our integrated FT sensor and would not be functional with other third party sensors. The purpose of the code is just like your description, to maintain a touching force in a specific direction. 

If using our integrated FT sensor under ROS environment, there will be a topic publishing the filtered and compensated sensor reading data, which is  `/xarm/uf_ftsensor_ext_states`. Otherwise, you may use a third party FT sensor and publish the corresponding data by yourself.

SyllogismRXS, 2023-08-29:
We are going to be purchasing the xArm6 with xArm's external Force/Torque (FT) sensor. Are there any examples of finding the touch point with that hardware configuration? Thanks!

penglongxiang, 2023-08-30:
You may monitor the data while command the robot to move along z- direction for example, and stop the robot when the Z force data exceed some threshold. Are you going to use ROS2 to realize this function?

SyllogismRXS, 2023-08-30:
Thanks for your response. Yes, we have a lot of experience with ROS2 and MoveIt2, so we will be using it. Your proposed solution makes sense. We were thinking of a similar approach, but I was unsure about the delay in time between when the robot gripper comes into contact with the work surface, the force data is received by a ROS2 node monitoring the force data, and then the command being sent back to the robot arm to stop moving down. Are there any parameters in the xArm6's firmware that we can tune to minimize the wear on the robot joints to account for that time delay? Thanks!

penglongxiang, 2023-08-31:
According to your description, I would probably suggest you to try the "**impedance control**" function with our integrated FT sensor, this function will make the robot tool side behave like a spring while in contact with the obstacle. Please first read (explanations available inside) and test the code from [8002](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/8002-impedance.py) example of SDK repository, the parameters worth tuning would be the equivalent stiffness (`K_pos`/`K_ori`), mass(`M`)/moment of inertia(`J`), and the impedance enabling direction (`c_axis`).  Maybe just tuning the stiffness would be enough, and the M/J parameter should not be too small for stability consideration.

Note: if you are attaching gripper or not just operating toward Z direction, please do a sensor/load compensation first. Run [example 8004](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/8004-load_identify.py) or do it in UFACTORY Studio. 

Finally, to implement it in ROS2, please follow ReadMe [chapter 5.4](https://github.com/xArm-Developer/xarm_ros2/tree/humble#54-xarm_api) to make the proper configuration to enable the relevant services. The services are listed in a file for your reference:
[xarm_user_params.yaml.txt](https://github.com/xArm-Developer/xarm_ros2/files/12481840/xarm_user_params.yaml.txt)
And here are the ros2 service calling examples to realize the behaviors in SDK example 8001-8003:
[ft_sensor_ros2_cmd.txt](https://github.com/xArm-Developer/xarm_ros2/files/12481850/ft_sensor_ros2_cmd.txt)

SyllogismRXS, 2023-09-15:
Thank you for the information. I have successfully installed the force torque sensor on the xarm6 and I'm able to call the ros2 service to enable the force torque sensor with the command: `ros2 service call /xarm/ft_sensor_enable xarm_msgs/srv/SetInt16 "{data: 1"}`. After issuing the command, I can see valid data being published to `/xarm/uf_ftsensor_ext_states` and `/xarm/uf_ftsensor_raw_states`. However, when the FT sensor is enabled, I see the following ERROR output from the ros2_control_node: `[ros2_control_node-6] [ERROR] [1694793504.462182933] [UFACTORY.RobotHW]: [192.168.1.210] Robot State detected! State: 5`. This forces the ros2 control software into a mode where it won't send new joint commands to the xarm6. If I restart the initial ros2 launch command (`ros2 launch xarm_moveit_config xarm6_moveit_realmove.launch.py robot_ip:=192.168.1.210 report_type:=rich`), I can see the valid data being published to FT data topics and I can use MoveIt to send joint commands. Is this a bug in the xarm ros2_control hardware software or am I doing something in the wrong order? Thanks!

penglongxiang, 2023-09-18:
Hi, it is true that the ft_sensor_enable will automatically set the robot state to **5**, which is some kind of **reset state** and the arm will not be ready for new command execution until you manually set the state back to 0.  Please **upgrade the controller firmware to the latest** version and use `rosservice call /xarm/set_state 0` to set the robot to READY state after enabling the FT sensor.

SyllogismRXS, 2023-10-27:
Thanks for you help. We have the ft sensor working with the impedance controller now to find the table height.

closed 2023-10-27 by SyllogismRXS