xArm-Developer/xarm_ros2 #43: Unable to use the gripper when running Moveit2 to control a simulated lite6 robotic arm

GBW48, 2023-08-30:
Hi,
**I cannot use the gripper when running Moveit2 to control a simulated lite6 robotic arm.**

When I run ros2 "launch xarm_moveit_config lite6_moveit_fake.launch.py add_gripper:=true" the following happens:
- The lite6 arm and gripper are displayed
- I can only control the position of the arm using motion planning

However, my problem is that I cannot control the lite6 gripper.  Please note the following:
- The Planning Group (in the Planning tab of the MotionPlanning panel) only has one entry "lite6" for the entire lite6 robot arm.  There is no entry for the lite6 gripper.
- When I select "lite6" in the Planning Group, the Joints tab (in the MotionPlanning panel) only shows robot arm joints (joint1 thru joint6), no joints are shown for the gripper fingers

I believe my build of ROS2, Moveit2, Gazebo, Gazebo_ros_packages with the necessary uFactory xarm_ros2 files is complete and functional. When I run "ros2 launch xarm_moveit_config xarm6_moveit_fake.launch.py add_gripper:=true":
- The Planning Group (in the Planning tab of the MotionPlanning panel) has two entries "xarm6" and "xarm_gripper" for the entire xarm6 robot arm. 
- When I select "xarm_gripper" in the Planning Group, the Joints tab (in the MotionPlanning panel) shows "drive_joint", "left_finger_joint", "left_inner_knuckle", "right_inner_knuckle", "right_outer_knuckle", "right_finger_joint".

I am running Ubuntu 22.04 and have carefully followed the installation instructions as detailed in https://github.com/xArm-Developer/xarm_ros2/tree/humble:
- 3.1 Install ROS2
- 3.2 Install Moveit2
- 3.3 Install Gazebo
- 3.4 Install gazebo_ros_pkgs
- 4.1 Create a workspace
- 4.2 Obtain source code of "xarm_ros2" repository
- 4.3 Update "xarm_ros2" repository
- 4.4 Install dependencies
- 4.5 Build xarm_ros2

**Can you please explain why I am unable to control the gripper when running Moveit2 to control a simulated lite6 robotic arm?  Please provide the necessary steps to enable support for the lite6 gripper when running Moveit2 and Gazebo2.**

Thanks in advance for any help you can provide

vimior, 2023-08-31:
@GBW48 
Because the structure of Lite Gripper is different from that of XArm Gripper, it cannot be controlled through moveit, so it cannot be simulated. On the real machine, it can only be controlled through service.

GBW48, 2023-08-31:
This is disappointing and quite unexpected!

When I purchased my lit6 robotic arm I was told that there would be full support for the arm and gripper under ROS2, Moveit2 and Gazebo2.  Further, the documentation seemed to indicate that there was support because of the "add_gripper:=true" parameter in sections 5.6 (xarm_moveit_config) and 5.7 (xarm_planner) in the xarm documenation (https://github.com/xArm-Developer/xarm_ros2/tree/humble). Now, unfortunately, I understand that because of the "structure of the Lite Gripper" control of the gripper cannot be simulated."

Given this I have some questions:
1. Can you explain what  the "structure of the Lite Gripper" control of the gripper cannot be simulated." means?  I'm having trouble understanding why the lite6 gripper cannot  be implemented in Moveit2?
2. Are there any plans to provide such support in the future?  If so, when?  Otherwise, why are you unable to provide such support for the gripper or the vacuum griper for that matter?
3. In your reply you indicated "On the real machine, it can only be controlled through service." Does this mean that I can directly control a real (not simulated) lite6 gripper using Moveit2?  Specifically, I am concerned that I will only be able to control the gripper through the API of the SDK or the service of ROS2 and NOT through the API of the MoveIt2 API.  
4. If it is possible for Moveit2 to directly control the operation of a real gripper, please provide documentation on how this is accomplished. 
5. In section 5.7 (xarm_planner) of the xarm_ros2 documentation, are there equivalent lite6 Moveit2 commands, to the following xArm commands:
   - run test program（control through service）
     $ ros2 launch xarm_planner test_xarm_planner_client_joint.launch.py dof:=6
     $ ros2 launch xarm_planner test_xarm_planner_client_pose.launch.py dof:=6
   - run test program（control gripper through service）
     ros2 launch xarm_planner test_xarm_gripper_planner_client_joint.launch.py dof:=6

Again, thanks in advance for any help you can provide.

penglongxiang, 2023-08-31:
Hi @GBW48, here are the responses for your questions:
1. _Can you explain what the "structure of the Lite Gripper" control of the gripper cannot be simulated." means? I'm having trouble understanding why the lite6 gripper cannot be implemented in Moveit2?_
- The mechanical design of Lite6 gripper and xArm Gripper are quite different. The latter has position feedback and is controlled by rotational motor and can achieve position and velocity control, while the former is controlled by a vacuum pump and only have binary states (open or close), it has no position feedback so there is no way for motion plan for Lite 6 gripper.

2. _Are there any plans to provide such support in the future? If so, when? Otherwise, why are you unable to provide such support for the gripper or the vacuum griper for that matter?_
- Since the Lite gripper is like a binary state IO device, it may be difficult to do a fake planning and control by moveit2. However, we will try to see what can be done to support its simulation, we can not tell whether it can be successful or when it will be implemented for now. 
- Meanwhile, you can see if there is anyway to attach your object in the planning scene directly to the gripper and disable the collision between them when you simulate the grasp operation. I think there are relevant APIs in Moveit, you can check something like `moveit_msgs::AttachedCollisionObject`.

3. _In your reply you indicated "On the real machine, it can only be controlled through service." Does this mean that I can directly control a real (not simulated) lite6 gripper using Moveit2? Specifically, I am concerned that I will only be able to control the gripper through the API of the SDK or the service of ROS2 and NOT through the API of the MoveIt2 API._
- I believe in your ROS2 program code, you can use both Moveit2 API as well as calling our provided ROS2 service. The relevant service names are: `open_lite6_gripper`, `close_lite6_gripper` and `stop_lite6_gripper`.

4. _If it is possible for Moveit2 to directly control the operation of a real gripper, please provide documentation on how this is accomplished._ 
- For Lite6 gripper, Moveit2 can not make motion planning for it, as stated above. Please use the ROS service mentioned in last answer to control the real gripper.

5. - For now, there is no ready-to-use code for Lite 6 equivalent, since we expect users can read the provided xArm test code and write the Lite 6 version in the same way. We can add a Lite 6 version test code later, and we will add the open/close service calling of Lite 6 gripper for the real machine control.

vimior, 2023-09-01:
@GBW48 
Regarding the use of Lite Gripper (only in real environments), a new routine has been added. You can use it to get the latest code.
1. `ros2 launch xarm_planner lite6_planner_realmove.launch.py`
2. `ros2 launch xarm_planner test_lite_gripper_realmove.launch.py`

GBW48, 2023-09-04:
I appreciate that you responded so quickly to my problem.  

I tried your suggested commands with my lite6.  Unfortunately, they did not work.  Both commands seemed to execute correctly, but the gripper fingers did not open or close on the real lite6.

Working from a previous build that included a basic (and fully updated) version of Ubuntu 22.04, ROS2, MoveIt2, Gazebo2, and gazebo_ros_pkgs, I followed the installation instructions as detailed in https://github.com/xArm-Developer/xarm_ros2/tree/humble:
- 4.1 Create a workspace
- 4.2 Obtain source code of the "xarm_ros2" repository
- 4.3 Update "xarm_ros2" repository
- 4.4 Install dependencies
- 4.5 Build xarm_ros2

Here are the results when I ran the commands you provided:
## Command 1	
> ros2 launch xarm_planner lite6_planner_realmove.launch.py robot_ip:=192.168.15.240 add_gripper:=true
### Command 1 observed results
- The lite6 arm and lite6 gripper was displayed in Moveit2
- Could control the position of the arm using Moveit2 motion planning
- The Planning Group only showed the "lite6" arm, there was no way to control the gripper 
### Command 1 results
- Refer to  Command_1_results.txt
### Command 1 log
- Refer to  Command_1_log.txt

## Command 2 (run in another terminal)
> ros2 launch xarm_planner test_lite_gripper_realmove.launch.py
### Command 2 observed results
- There was no change in the position of the lite6 arm or gripper
- No movement of the gripper fingers was observed
### Command 2 results
- Refer to  Command_2_results.txt
### Command 2 log
- Refer to  Command_2_log.txt

Please review the attached files and let me know why I am still unable to open and close the gripper on my lite6 robot.

Again, thanks in advance for any help you can provide.
[Command_1_results.txt](https://github.com/xArm-Developer/xarm_ros2/files/12509719/Command_1_results.txt)
[Command_2_log.txt](https://github.com/xArm-Developer/xarm_ros2/files/12509720/Command_2_log.txt)
[Command_2_results.txt](https://github.com/xArm-Developer/xarm_ros2/files/12509721/Command_2_results.txt)
[Command_1_log.txt](https://github.com/xArm-Developer/xarm_ros2/files/12509722/Command_1_log.txt)

vimior, 2023-09-04:
@GBW48 
Sorry, I forgot to tell you that the service controlling Lite6 is not enabled by default. You need to modify the configuration to enable it on demand.
For configuration modifications, please refer to [ReadMe](https://github.com/xArm-Developer/xarm_ros2/tree/humble) 5.4. The relevant service names are: `open_lite6_gripper`, `close_lite6_gripper` and `stop_lite6_gripper`.

example (*xarm_api/config/xarm_user_params.yaml*):
```
ufactory_driver:
  ros__parameters:
    services:
      open_lite6_gripper: true
      close_lite6_gripper: true
      stop_lite6_gripper: true
```

GBW48, 2023-09-04:
I tried your suggested edits to the xarm_params.yaml file in the ~/dev_ws/src/xarm_ros2/xarm_api/config directory.  Please refer to the modified yaml file in the attachments. 

Unfortunately, these changes did not work. Again, even after restarting Ubuntu, both commands seemed to execute correctly, but the gripper fingers did not open or close on the real lite6.

Here are the results when I ran the commands you provided:
## Command 1	
> ros2 launch xarm_planner lite6_planner_realmove.launch.py robot_ip:=192.168.15.240 add_gripper:=true
### Command 1 observed results
- The lite6 arm and lite6 gripper was displayed in Moveit2
- Could control the position of the arm using Moveit2 motion planning
- The Planning Group only showed the "lite6" arm, there was no way to control the gripper 
### Command 1 results
- Refer to  Command_1_results.txt
### Command 1 log
- Refer to  Command_1_log.txt

## Command 2 (run in another terminal)
> ros2 launch xarm_planner test_lite_gripper_realmove.launch.py
### Command 2 observed results
- There was no change in the position of the lite6 arm or gripper
- No movement of the gripper fingers was observed
### Command 2 results
- Refer to  Command_2_results.txt
### Command 2 log
- Refer to  Command_2_log.txt

Please review the attached files and let me know why I am still unable to open and close the gripper on my lite6 robot.

Again, thanks in advance for any help you can provide.
[xarm_params.yaml.txt](https://github.com/xArm-Developer/xarm_ros2/files/12517153/xarm_params.yaml.txt)
[Command_1_result.txt](https://github.com/xArm-Developer/xarm_ros2/files/12517154/Command_1_result.txt)
[Command_1_log.txt](https://github.com/xArm-Developer/xarm_ros2/files/12517155/Command_1_log.txt)
[Command_2_result.txt](https://github.com/xArm-Developer/xarm_ros2/files/12517156/Command_2_result.txt)
[Command_2_log.txt](https://github.com/xArm-Developer/xarm_ros2/files/12517157/Command_2_log.txt)

vimior, 2023-09-05:
@GBW48 
Have you recompiled after modifying the configuration? The default method of ROS2 needs to be recompiled to be effective.
The second command is actually just a simple client, calling /ufactory/open_lite6_gripper and /ufactory/close_lite6_gripper. I looked at the log and saw that it was waiting to connect to the service, so you can check all services after starting the first command.

GBW48, 2023-09-05:
I did not do anything but edit the yaml file.  I assume that by  "recompiled" you want me to rebuild my lite6 ROS/Moveit1 environment by going through the following steps?
- 4.1 Create a workspace
- 4.2 Obtain source code of the "xarm_ros2" repository
- 4.3 Update "xarm_ros2" repository
- 4.4 Install dependencies
- 4.5 Build xarm_ros2

Please let me know if you want me to perform these steps and anything else.

With respect to your comment about the second command.  Do you still want me to run the second command?

Also, how do I check that the gripper services are running?

Thanks for all your help!
...Gary

vimior, 2023-09-05:
@GBW48 
1. Recompiling only requires 4.5, i.e. `colcon build`
2. To check all services, run `ros2 service list` in another terminal to check all services.

You can try recompiling and testing first. If the second command is still waiting, then query all services. The services are created by the first command and have nothing to do with whether the second command is executed.

GBW48, 2023-09-05:
After following step 4.5 (https://github.com/xArm-Developer/xarm_ros2/tree/humble), and performing a colcon build,  I can now control the lite6 gripper as follows:

`launch xarm_planner test_lite_gripper_realmove.launch.py`
--The gripper repeatedly opens and closes
-- I have to Ctrl-C to stop the repeated opening and closing of the gripper

`ros2 service call /ufactory/open_lite6_gripper xarm_msgs/srv/Call`
-- The gripper opens

`ros2 service call /ufactory/close_lite6_gripper xarm_msgs/srv/Call`
-- The gripper closes

`ros2 service call /ufactory/stop_lite6_gripper xarm_msgs/srv/Call`
-- The gripper stops operating

In addition, when I run `ros2 service list` the following gripper services are returned: /ufactory/open_lite6_gripper, /ufactory/close_lite6_gripper, and /ufactory/stop_lite6_gripper.

So at this point, when running MoveIt2, I can control the operation of the real gripper.

**The only remaining problem is that the gripper fingers in Moveit do not open and close in response to ROS2 service commands. In this case, the gripper fingers work on the real robot, but do not work on the gripper displayed by Moveit.**

Again, thanks for all your help!

vimior, 2023-09-05:
@GBW48 
This question goes back to the original explanation. Due to structural problems, the driving method is different, and there is no position feedback. The model is only a whole, so the Lite Gripper model can only be displayed statically.