xArm-Developer/xarm_ros2 #49: Speed Limit Exceeded - Lite 6

peterdavidfagan, 2023-10-16:
I am currently facing issues when executing planned trajectories using the humble branch. This is the first time I refreshed my lite6 workspace in a while, is this a known issue?

```
[ros2_control_node-6] [INFO] [1697471980.951808737] [lite6_traj_controller]: Received new action goal
[ros2_control_node-6] [INFO] [1697471980.951906433] [lite6_traj_controller]: Accepted new action goal
[motion_planning_python_api_tutorial.py-1] [INFO] [1697471980.952342756] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: lite6_traj_controller started execution
[motion_planning_python_api_tutorial.py-1] [INFO] [1697471980.952367408] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-6] [WARN] [1697471981.070178757] [UFACTORY.RobotHW]: [192.168.1.156] set_servo_angle_j, ret= 1
[ros2_control_node-6] [WARN] [1697471981.090329883] [UFACTORY.RobotHW]: [192.168.1.156] set_servo_angle_j, ret= 1
[ros2_control_node-6] [WARN] [1697471981.110257122] [UFACTORY.RobotHW]: [192.168.1.156] set_servo_angle_j, ret= 1
[ros2_control_node-6] [WARN] [1697471981.130212239] [UFACTORY.RobotHW]: [192.168.1.156] set_servo_angle_j, ret= 1
[ros2_control_node-6] [WARN] [1697471981.150298455] [UFACTORY.RobotHW]: [192.168.1.156] set_servo_angle_j, ret= 1
[ros2_control_node-6] [ERROR] [1697471981.168882239] [UFACTORY.RobotHW]: [192.168.1.156] UFACTORY Error detected! Code C24 -> [ Speed Exceeds Limit ] 
```

peterdavidfagan, 2023-10-17:
Looks like this is a known issue as referenced [here](http://help.ufactory.cc/en/articles/4712781-guide-user-to-solve-the-error-c24-speed-exceeds-limit). 

I'll run some more tests to understand why it is occuring.

penglongxiang, 2023-10-17:
Hi @peterdavidfagan, it is better to reproduce this issue and download the log package immediately after that. We will analyze and find out the cause based on your provided log.

The log can be downloaded from UFACTORY studio: Settings->System->Log->Download (v1.x) or  Settings->My Device->Log->Download (v2.x).

peterdavidfagan, 2023-10-17:
Thanks @penglongxiang, ok I am going to try to alter a few settings and if I cannot resolve it myself I will send these logs as requested. 

Right now I am guessing that altering my device settings from the browser may help as I assume the driver is configured using these? 

Are the joint limits [here](https://github.com/xArm-Developer/xarm_ros2/blob/humble/xarm_moveit_config/config/lite6/joint_limits.yaml) tested?

peterdavidfagan, 2023-10-17:
**Log File**
[LI100510221B20_2023-10-17_11-10-47.tar.gz](https://github.com/xArm-Developer/xarm_ros2/files/12927417/LI100510221B20_2023-10-17_11-10-47.tar.gz)

penglongxiang, 2023-10-17:
Thanks for the detailed information. We will read the log and try reproducing this issue on our side. 

I guess this is related to a known issue: under external servo mode (1), we added linear speed monitoring and the limit is 500mm/s for Lite6, but sometimes the linear speed estimation maybe inaccurate and thus false triggering this error. We have optimized this estimation and make the speed monitor threshold configurable in the next version of controller firmware.  

If this is an urgent issue for your development, we can provide a beta version for you, which temporally disabled this linear speed monitoring in servo mode. I will update with the analyze result tomorrow, thank you.

peterdavidfagan, 2023-10-17:
> I guess this is related to a known issue: under external servo mode (1), we added linear speed monitoring and the limit is 500mm/s for Lite6, but sometimes the linear speed estimation maybe [...]

Thanks for the explanation, so is my understanding correct in that the driver essentially is having issues with controlling joint speed based on the position commands passed by ROS 2 control. So ROS 2 being the external servo command? How is the estimation currently implemented in the driver?

> If this is an urgent issue for your development, we can provide a beta version for you, which temporally disabled this linear speed monitoring in servo mode. I will update with the analyze result tomorrow, thank you.

I am happy to test the beta on my system. It would be very helpful as I plan to use this robot to test changes to MoveIt and also to prototype research experiments. I am happy to provide swift feedback.

penglongxiang, 2023-10-17:
Yes, in your application ROS2 controller operates as external servo command generator, our controller firmware receives the high-frequency command and executes in real-time, however since the received external servo command is asynchronous with our internal control frequency, there are chances that the command speed estimation is incorrect. 

Cool, if beta version is OK for you, we will provide it once it is confirmed to be this reason, and it will have to wait until tomorrow.

peterdavidfagan, 2023-10-17:
> Yes, in your application ROS2 controller operates as external servo command generator, our controller firmware receives the high-frequency command and executes in real-time, however since the [...]

If I deploy with my ROS 2 control running on a RT patched NUC should this issue resolve itself? Right now I am prototyping on a workstation without RT patch. 

> Cool, if beta version is OK for you, we will provide it once it is confirmed to be this reason, and it will have to wait until tomorrow.

Fantastic, thanks for the awesome support on this @penglongxiang.

penglongxiang, 2023-10-18:
Hi @peterdavidfagan, Thanks for the further information. The real cause of Overspeed Error may not be the speed estimation I mentioned yesterday. In your log we found something strange, our controller box did receive all-zero command from the upper ROS2 controller application, to be more specific, the generated 50Hz command trajectory always seems to be starting from home (zero joint) position. 

However, we have tested and confirmed that our provided Moveit demo application (humble branch in ROS2 humble, and rolling branch in ROS2 rolling, which may not be the latest) would not behave like this. Are you using a customized or development version of ROS2 or Moveit2? Would you please check on your side for potential reason causing this issue, like the robot start state did not get updated to ROS2 controller somehow? 

Further more, would you be able to reproduce this phenomenon just running `ros2 launch xarm_moveit_config lite6_moveit_realmove.launch.py robot_ip:=<your_robot_ip>`  and controlling the lite6 in the GUI?

peterdavidfagan, 2023-10-18:
Thanks @penglongxiang,

> In your log we found something strange, our controller box did receive all-zero command from the upper ROS2 controller application, to be more specific, the generated 50Hz command trajectory always seems to be starting from home (zero joint) position.

I'll investigate the robot state publisher now to ensure it is working and confirm once resolved. I'll close this issue once I have a full explanation on what caused the issue.

peterdavidfagan, 2023-10-18:
The issue was on my side, my joint state publisher attempted to map `ufactory/joint_states`, the namespace was updated in more recent changes to the repository to `xarm/joint_states`. Thanks @penglongxiang for the support on this.

closed 2023-10-18 by peterdavidfagan

penglongxiang, 2023-10-19:
No problem, glad to know the issue was solved.