xArm-Developer/xarm_ros2 #50: robot_state_publisher exit code -15 during launch xarm7 description and Moveit config

Royo666, 2023-10-20:
Hi,

When I tried to run the package to view the xarm7 description, the error related to robot_state_publisher keeps coming out. Please see the detail in the screenshot below: 
[image]

I ran the rqt to check if there is any fatal error, but there was not:

[image]

So I tried to run the xarm_moveit_config package to see if there was same issue with call out the robot_state_publisher node. I thought there was the same issue as I had when I tried to call out the description node. Please see the detail below:
[image]

Is it due to the ID problem (since I haven't change anything related to ROS Domain)? Or it is something else.

Thank you

Best wishes

Royo666, 2023-10-20:
Sorry for the limited detail description for the issue. Here is the additional description for the issue.

When I tried to run the xarm_description package, the terminal shows:

```
/opt/ros/foxy/bin/ros2:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
[INFO] [launch]: All log files can be found below /home/zo22/.ros/log/2023-10-20-14-24-00-241568-sie187-lap-4519
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=PackageNotFoundError("package 'joint_state_publisher' not found, searching: ['/home/zo22/xarm_ws/install/xarm_planner', '/home/zo22/xarm_ws/install/xarm_moveit_servo', '/home/zo22/xarm_ws/install/xarm_moveit_config', '/home/zo22/xarm_ws/install/xarm_gazebo', '/home/zo22/xarm_ws/install/xarm_controller', '/home/zo22/xarm_ws/install/xarm_api', '/home/zo22/xarm_ws/install/xarm_sdk', '/home/zo22/xarm_ws/install/xarm_msgs', '/home/zo22/xarm_ws/install/xarm_description', '/home/zo22/xarm_ws/install/realsense_gazebo_plugin', '/opt/ros/foxy']")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/ament_index_python/packages.py", line 50, in get_package_prefix
    content, package_prefix = get_resource('packages', package_name)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ament_index_python/resources.py", line 48, in get_resource
    raise LookupError(
LookupError: Could not find the resource 'joint_state_publisher' of type 'packages'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 6 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 453, in execute
    ret = super().execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 823, in execute
    self.__expand_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 668, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 668, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/substitutions/executable_in_package.py", line 76, in perform
    package_prefix = super().perform(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/substitutions/find_package.py", line 79, in perform
    result = self.find(package)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/substitutions/find_package.py", line 96, in find
    return get_package_prefix(package_name)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ament_index_python/packages.py", line 52, in get_package_prefix
    raise PackageNotFoundError(
ament_index_python.packages.PackageNotFoundError: "package 'joint_state_publisher' not found, searching: ['/home/zo22/xarm_ws/install/xarm_planner', '/home/zo22/xarm_ws/install/xarm_moveit_servo', '/home/zo22/xarm_ws/install/xarm_moveit_config', '/home/zo22/xarm_ws/install/xarm_gazebo', '/home/zo22/xarm_ws/install/xarm_controller', '/home/zo22/xarm_ws/install/xarm_api', '/home/zo22/xarm_ws/install/xa [... cut ...]
[ERROR] [robot_state_publisher-1]: process[robot_state_publisher-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [robot_state_publisher-1]: sending signal 'SIGTERM' to process[robot_state_publisher-1]
[ERROR] [robot_state_publisher-1]: process has died [pid 4522, exit code -15, cmd '/opt/ros/foxy/lib/robot_state_publisher/robot_state_publisher --ros-args --params-file /tmp/launch_params_lwz0hwdj -r /tf:=tf -r /tf_static:=tf_static'].
```
I think what happens here is that the robot_state_publisher node is executed, but is  crashed for an unknown reason.

vimior, 2023-10-23:
@Royo666 
After looking at the terminal output, it seems that the dependency `joint_state_publisher` is missing. You can try to completely recompile the dependency installation and then use it.

closed 2024-05-27 by Royo666