xArm-Developer/xarm_ros2 #51: controllers.yaml and fake_controllers.yaml are not being used

DaniGarciaLopez, 2023-10-22:
I've noticed that the files `controllers.yaml` and `fake_controllers.yaml` under `xarm_moveit_config/config/{robot}` are no longer in use. Currently, the parameters for `traj_controller` are sourced from `xarm_controller/config/{robot}_controllers.yaml`.

What's the point of keeping these files? They can only lead to confusion for new users attempting to modify these values but not seeing any effect.

vimior, 2023-10-23:
@DaniGarciaLopez 
`xarm_controller/config/{robot}_controllers.yaml` is used by __joint_trajectory_controller/JointTrajectoryController__.
`xarm_moveit_config/config/{robot}/controllers.yaml` is used by __moveit_simple_controller_manager/FollowJointTrajectoryHandle__.

DaniGarciaLopez, 2023-10-25:
Hi @vimior, in `xarm_ros`, I noticed that `xarm_moveit_config/config/{robot}/controllers.yaml` is actually being used. However, in `xarm_ros2`, I searched and found that only `xarm_controller/config/{robot}_controllers.yaml` is being sourced.

vimior, 2023-10-26:
@DaniGarciaLopez 
`xarm_moveit_config/config/{robot}/controllers.yaml` is also used in xarm_ros2. The parameters in this file are used when starting the `moveit_ros_move_group/move_group node`.
[_robot_moveit_common.launch.py](https://github.com/xArm-Developer/xarm_ros2/blob/humble/xarm_moveit_config/launch/_robot_moveit_common.launch.py)

DaniGarciaLopez, 2023-10-26:
Ok, I see now the reference to the file [here](https://github.com/xArm-Developer/xarm_ros2/blob/fcbefc421468487b1130196c67b4a1f4326d57e2/xarm_moveit_config/launch/_robot_moveit_common.launch.py#L116C84-L116C91). I was trying to look for the full name of the file and that's why I could't find it. Thanks @vimior!

closed 2023-10-26 by DaniGarciaLopez