xArm-Developer/xarm_ros2 #57: cant interface with dual_xarm6_moveit_fake.launch.py

DailyNir, 2023-11-20:
i am trying to connect the move groups L_xarm6 and R_xarm6 to send them goal points, and plan and execute through code  and it is proving fairly difficult and almost imposibble. 
using ROS2 Humble, please advise for a tutorial or an example code for how to interface with it so i can plan within the rviz environment butthrough a package i write.

thank you.

vimior, 2023-11-22:
@DailyNir 
In order to control dual robot arms in __xarm_planner__, I have added an [example](https://github.com/xArm-Developer/xarm_ros2/blob/humble/xarm_planner/launch/test_dual_xarm_planner_api_joint.launch.py) for you to refer to. Please get the latest code and recompile it.

DailyNir, 2023-11-29:
thanks for your code, works great. can i create 2 different move_groups for each arm in order to control both simultaneously ? moreover can i plan in the same scene for both arm  simultaneously if i split too 2 move_groups?