xArm-Developer/xarm_ros2 #60: ROS2 moveit planners, set planning time in code

DailyNir, 2024-01-08:
Hello xArm Developer Team,

I am currently working with the xArm ROS 2 wrapper and I've encountered a scenario where I need to customize the planning time for the robot's movements. However, I am having difficulty locating the specific part of the code or the parameter that allows me to modify this setting.

Could you please guide me on how to change the planning time for the xArm in the ROS 2 wrapper? Any documentation or examples related to this would be highly appreciated.

Additionally, if this feature is not currently supported, I would like to suggest it as a potential enhancement for future releases. It would be incredibly beneficial for users requiring more flexibility in planning complex movements.

Thank you for your time and assistance.

Best regards

penglongxiang, 2024-01-09:
Hi @DailyNir, could you please elaborate on " **the planning time for robot's movement**" ? Does that mean the planner calculation time or the trajectory execution time? Moreover, are you using Moveit or the wrapped ros service for robot motion planning?  

Currently we do not have an effective interface to specify the execution time for a planned motion, only target speed and acceleration can be specified.

DailyNir, 2024-01-09:
Well I was talking about the time it takes a planner to find a trajectory, I found that i can use move_group_.setPlanningTime function in the xarm_planner.cpp file but the planners are still failing, do you have any suggestions for choosing the right planner or getting another one into OMPL? I have them 30 seconded and 20 attempts and still failing to find a correct trajectory .

penglongxiang, 2024-01-09:
`xarm_planner` is actually using Moveit planning interface in the background, it is from third party and not under our control. May I know which robot model you are using? Would you please share the initial joint configuration and command target when the planning fails? 

You may go to **xarm_moveit_config/config/<your_robot_model>/ompl_planning.yaml** to configure the `default_planner_config` and the corresponding planner parameters if any.