xArm-Developer/xarm_ros2 #61: ROS2 Humble gazebo package installation

camroots, 2024-01-12:
Hi, I have a LITE 6 arm that I am looking to program. First I would like to simulate the arm before interacting with the hardware.

I have an Mac M1, running Parallels 19 that has an Ubuntu 22.04.2 ARM64 virtual machine. I am trying to follow the installation instructions [here](https://github.com/xArm-Developer/xarm_ros2/tree/humble) for installing ROS2 Humble. 

## Installing gazebo
The readme points to [here](https://classic.gazebosim.org/tutorials?tut=install_ubuntu) for installing gazebo with the command `curl -sSL http://get.gazebosim.org | sh`. This results in the following error: 

```
E: Package 'libgazebo-dev' has no installation candidate
E: Package 'gazebo' has no installation candidate
```

## Installing gazebo_ros_pkgs
The readme makes reference to [gazebo_ros_pkgs](https://classic.gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros) page which contains installation instructions for installing the packages for Foxy rather than Humble. 

I considered attempting the Foxy installation but when running `sudo apt install ros-foxy-gazebo-ros-pkgs` I am provided with:
```
E: Unable to locate package ros-foxy-gazebo-ros-pkgs
```

## Fortress installation
If I follow the instructions on the Gazebo website I am able to install the fortress version of gazebo and am able to run `shapes.sdf`.

The issue with this though is when I attempt to install the xArm dependencies ([as per here](https://github.com/xArm-Developer/xarm_ros2/tree/humble?tab=readme-ov-file#44-install-dependencies)) I receive the following error:

```
executing command [sudo -H apt-get install -y ros-humble-gazebo-plugins]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-gazebo-plugins
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-humble-gazebo-plugins] failed

```

Would someone be able to provide some guidance on which packages I should be installing? Would there be any expected issues running with my M1 Parallels setup?

Thank you.

penglongxiang, 2024-01-18:
Sorry we do not have the experience of M1 Mac platform installation, and we are not sure about the compatibility of the packages needed. Will leave this issue open for other users' help.

cemoke, 2024-09-08:
I am having the same issue. Is it possible to use the repo without having the gazebo? I will not use the simulation; I will use the robot and Rviz. I think it should be possible to build without the Gazebo

zongguangliu, 2025-06-11:
it is possible to use it without gazebo, in the src folder -> xarm_gazebo
add a empty file name COLCON_IGNORE
and also in the src folder -> thirdparty -> realsense_gazebo_plugin
add a empty file name COLCON_IGNORE