xArm-Developer/xarm_ros2 #62: Lite 6 Impedence Control

aoberai, 2024-01-18:
Aside from the backdrivability of the 3 joints near the eef of the lite6, there also appears to be a 6 axis force sensor (as is demonstrated in the existence of a manual mode for the lite6). Why can't that be used for impedance control? This would preferably exist firmware side which has a higher control frequency. Why does the xarm7 require a separate force sensor instead of the one used for manual mode?

Also, can I get passive compliance on the back drivable joints of the lite6 by setting torque / current limits for specifically those joints? In my case, this would be favorable to the current collision detection fault that is sent instead (that requires a restart of the arm).

penglongxiang, 2024-01-18:
Hi @aoberai, there is no integrated force/torque (F/T) sensor INSIDE any of our robotic arm. The collision detection and manual mode is achieved by **motor current** observation and robot dynamic modeling. Since there is no direct torque sensor at the joint output, we do not provide current control mode to general users, it would not be safe for use without the exact friction model of the gear transmission (unpredictable at static state). 

We provide a software integrated third party 6D F/T sensor for xArm Series, it can be added on the tool-side and be used for admittance control. However, it can not be used on Lite6 because the hardware weight will exceed the capacity of it.

aoberai, 2024-01-18:
Is the manual mode motor current observations done just off of the 3 back drivable joints because the nonbackdrivable harmonic joints I presume won't give you even approximate torque readings (because no current is required to maintain joint state)?

penglongxiang, 2024-01-18:
The current observation is done for ALL the joints. We will compensate the friction when the joint is moving with an identified model, it is not precise but enough for collision detection and manual mode.

aoberai, 2024-01-18:
What would you recommend I do if I want compliant control on the Lite6?