xArm-Developer/xarm_ros2 #63: Request: UFACTORY BIO Gripper

J-Burgess, 2024-01-29:
Hi, thank you for the useful repository! 

I am writing to request the addition of the UFACTORY BIO Gripper to the repository. We are establishing a new children's disease research lab with two xARM 7's each with a BIO Gripper. Hoping to be able to simulate all trajectories and gripping of a sample plate to transfer between machines and then execute on hardware. I have looked around for the .xacro files but no luck so far on the internet! I will also try to get the ball rolling on adding this to the repository myself but I am new to the robotics scene! 

Thanks!

penglongxiang, 2024-01-30:
Hi, thanks for your complements. Sorry for not supporting BIO gripper in ROS/ROS2 so far, we will see how fast we can add it. By the way, there is actually no trajectory planning or position control for the real BIO gripper, it is simply controlled by a binary Open/Close command.

penglongxiang, 2024-02-28:
Hi @J-Burgess, we have updated the repository and added support for BIO gripper model, you can try launching the simulation by adding "add_bio_gripper:=true" argument.

closed 2024-04-03 by J-Burgess