xArm-Developer/xarm_ros2 #67: expected [string_array] got [string] - Move_group

Ep3896, 2024-04-05:
Good evening!
I'm using ROS2 Humble and I'm working on a Lite6 project.
After installing Moveit2 from source and right after launching:
>ros2 launch xarm_planner lite6_planner_fake.launch.py

 I've noticed that the move_group node dies and so i can't perform the motion planning on my robot. 
 
[image]

I was wondering what could I do to fix the error.
I've check on the moveit documentiation and see that:

"Warning: Most features in MoveIt will not work properly since additional parameters are required for full Move Group functionality. For a full setup, please continue with the [Move Group C++ Interface Tutorial]"

If anyone could give me advice on how to go through , I would be very pleased!

Thank you in advance,
Enrico

gaspatxo, 2024-04-05:
1. Make sure you follow the preparation and installation instructions from the [xarm_ros2 readme file](https://github.com/xArm-Developer/xarm_ros2/tree/humble#xarm_ros2) strictly
2. I had to **reboot** before being able to use the package (It could be the same issue but I do not remember)

Ep3896, 2024-04-05:
I can actually use the package, the problem is that the parameters are not correct, I think that there are parameters that need to be set or unset in order to let the program correctly go through. Maybe it is just an installation mistake but i really doubt about it

gaspatxo, 2024-04-06:
You are using Ubuntu 22.04, right? Is it a fresh installation or have you used it for other developing purposes?

Did you succesfully run the moveit2 tutorials?

Did you try using the package post-reboot after the installation session?

I ran the command on my setup, you can check the [CLI output](https://github.com/xArm-Developer/xarm_ros2/files/14892881/ros2-launch-xarm_planner-lite6_planner_fake-output.txt). As you can see the error you mention is not there.

Ep3896, 2024-04-06:
First of all, I appreciate your help! :D 
I think there are problems with the planning_pipeline. I don't know yet how to solve this problem but I think there are issues with passing parameters to move group. For parameters I mean the planners, because I see that where my program crushes, yours instead say:

>[move_group-4] [INFO] [1712390116.696721151] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'ompl'
>[move_group-4] [INFO] [1712390116.723966080] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'

So I really need to check how it is possible to fix that problem. Maybe it's a yaml problem? or maybe in the launch file? I am figuring out but if someone can help me I would really be grateful|

vimior, 2024-04-09:
@Ep3896 
Hello, could you provide the complete output of the terminal when running?

Ep3896, 2024-04-09:
Of course! this is the log error: [here](https://github.com/xArm-Developer/xarm_ros2/files/14916762/log.txt)

As for Rviz, this is the log on the window:

[image]

So the issue, I suppose, is that the planning pipeline yaml is not correct, but this is just my interpretation. 
I really trying to understand how to fix that but without great outcomes yet.

vimior, 2024-04-09:
@Ep3896 
I compared it and found that you did not load the output related to planning_pipeline and the abnormality is that the type of a certain configuration parameter is wrong. Have you modified anything?

Ep3896, 2024-04-09:
I don't think so!
However, I was able to point out where the program crashes:
> ros2 launch xarm_moveit_config _robot_moveit_common.launch.py
this launche file gives me the error but I'm trying to figure out why

and my

vimior, 2024-04-10:
@Ep3896 
Running `_robot_moveit_common.launch.py` alone is incomplete. You can run `lite6_moveit_fake.launch.py` to find the problem.
> ros2 launch xarm_moveit_config lite6_moveit_fake.launch.py

Ep3896, 2024-04-10:
yes! but I have already run :
> ros2 launch xarm_moveit_config lite6_moveit_fake.launch.py

But I have the exact same error!
Could it be libmoveit_planning_pipleline.so.2.9?
I'm running out of ideas, if you could provide to me, however your log of this:

>  ros2 launch xarm_moveit_config lite6_moveit_fake.launch.py

I would be really pleased :) 

Thank you in advance!

vimior, 2024-04-11:
@Ep3896 
This kind of missing libmoveit_planning_pipleline.so.2.9 is usually caused by incomplete system updates. You may need to obtain updates from a stable source.
In addition, the humble version here is libmoveit_planning_pipleline.so.2.5.5, and the rolling version is libmoveit_planning_pipleline.so.2.9.0

vimior, 2024-04-11:
@Ep3896
Maybe the problem lies in the libmoveit_planning_pipleline.so.2.9 version. This problem will also occur when I run it under the rolling version. I will look for the problem first.

vimior, 2024-04-11:
@Ep3896 
Hello, I looked through the moveit code and found that the configuration format has changed. Please refer to the [pr](https://github.com/ros-planning/moveit2/pull/2457) for details.

You can temporarily modify the `xarm_moveit_config/launch/_robot_moveit_common.launch.py` file to use, we will fix this problem later.
Search for the definition of `ompl_planning_pipeline_config`, and then modify it to the following format 
```
ompl_planning_pipeline_config = {
    'default_planning_pipeline': 'ompl',
    'planning_pipelines': ['ompl'],
    'ompl': {
        'planning_plugins': ['ompl_interface/OMPLPlanner'],
        'request_adapters': [
            'default_planning_request_adapters/ResolveConstraintFrames',
            'default_planning_request_adapters/ValidateWorkspaceBounds',
            'default_planning_request_adapters/CheckStartStateBounds',
            'default_planning_request_adapters/CheckStartStateCollision',
        ],
        'response_adapters': [
            'default_planning_response_adapters/AddTimeOptimalParameterization',
            'default_planning_response_adapters/ValidateSolution',
            'default_planning_response_adapters/DisplayMotionPath',
        ],
    }
}
```

Ep3896, 2024-04-11:
Thank you very much! 
I have learnt a lot :)
Greetings!

Ep3896, 2024-04-11:
I have installed Moveit2 from the main branch because they textually say:

>Currently the [main branch](https://github.com/ros-planning/moveit2) supports Rolling, Iron, and Humble. However, since it’s used for latest development, the main branch is unstable. For stable versions, please use the distro branches [humble](https://github.com/ros-planning/moveit2/tree/humble), or [iron](https://github.com/ros-planning/moveit2/tree/iron).

And my goal is to actually keep up with the latest developments, even if they are unstable.

Btw, Thank you very much for helping me out!

closed 2024-04-19 by Ep3896