xArm-Developer/xarm_ros2 #69: Question: what's the update rate at which "vc_set_joint_velocity" can be used?

vmayoral, 2024-04-25:
Hey there,

Been looking at `xarm_controller` hardware interface and noted that depending on `velocity_control_`, it uses either:
- [`vc_set_joint_velocity`](https://github.com/xArm-Developer/xarm_ros2/blob/master/xarm_controller/src/hardware/uf_robot_system_hardware.cpp#L395) or
-  [`set_servo_angle_j`](https://github.com/xArm-Developer/xarm_ros2/blob/master/xarm_controller/src/hardware/uf_robot_system_hardware.cpp#L407)

The update rate supported by `set_servo_angle_j` is well documented in the User Manual:

[image]

The `vc_set_joint_velocity` mode is however very little documented:

[image]

Could you clarify what's the supported update rate for the `vc_set_joint_velocity` mode for the `xArm 5` and `Lite 6` manipulators?

Thanks!

vmayoral, 2024-04-26:
Thanks @vimior! Do you happen to know if this also applies to the `Lite 6`? As compute capability is limited, I'm curious on whether this update rate is also confirmed there.

vimior, 2024-04-29:
@vmayoral 
It is the same as xarm, the maximum frequency is also 250HZ

closed 2024-04-29 by penglongxiang