xArm-Developer/xarm_ros2 #75: Testing MoveIt2 Python Bindings with real xArm6

adamanov, 2024-06-07:
Hello I am trying to use a moveit2 python biding on xarm6, and run a simple [example motion_planning_python_api_tutorial.py](https://github.com/moveit/moveit2_tutorials/blob/main/doc/examples/motion_planning_python_api/scripts/motion_planning_python_api_tutorial.py)`

By executing `ros2 launch xarm_moveit_config xarm6_moveit_fake.launch.py` and then the python example,  but getting to bellow error

```
/bin/python3 /home/lenova/robolab_ws/motion_planning_python_api_tutorial.py
[INFO] [1717775362.419700340] [moveit_1407267450.moveit.py.cpp_initializer]: Initialize rclcpp
[INFO] [1717775362.419740083] [moveit_1407267450.moveit.py.cpp_initializer]: Initialize node parameters
[INFO] [1717775362.419747302] [moveit_1407267450.moveit.py.cpp_initializer]: Initialize node and executor
[INFO] [1717775362.424865508] [moveit_1407267450.moveit.py.cpp_initializer]: Spin separate thread
[INFO] [1717775363.431238829] [moveit_1407267450.moveit.ros.rdf_loader]: Loaded robot model in 1.00628 seconds
[INFO] [1717775363.431274239] [moveit_1407267450.moveit.core.robot_model]: Loading robot model 'UF_ROBOT'...
[INFO] [1717775363.431282540] [moveit_1407267450.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[WARN] [1717775363.475375072] [moveit_1407267450.moveit.ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[INFO] [1717775363.606687157] [moveit_1407267450.moveit.ros.moveit_cpp]: Listening to 'joint_states' for joint states
[INFO] [1717775363.607611874] [moveit_1407267450.moveit.ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[INFO] [1717775363.609904673] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[INFO] [1717775363.610806049] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[INFO] [1717775363.610965652] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Starting planning scene monitor
[INFO] [1717775363.611344917] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Listening to '/planning_scene'
[INFO] [1717775363.611355517] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[INFO] [1717775363.611605042] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Listening to 'collision_object'
[INFO] [1717775363.611804670] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[WARN] [1717775363.612992705] [moveit_1407267450.moveit.ros.occupancy_map_monitor]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[ERROR] [1717775363.613011853] [moveit_1407267450.moveit.ros.occupancy_map_monitor]: No 3D sensor plugin(s) defined for octomap updates
[ERROR] [1717775363.689677223] [moveit_1407267450.moveit.ros.moveit_cpp]: Failed to load any planning pipelines.
[FATAL] [1717775363.689721237] [moveit_1407267450.moveit.ros.moveit_cpp]: Failed to load planning pipelines from parameter server
[INFO] [1717775363.689883009] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Stopped publishing maintained planning scene.
[INFO] [1717775363.689909254] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Stopping world geometry monitor
[INFO] [1717775363.689915990] [moveit_1407267450.moveit.ros.planning_scene_monitor]: Stopping planning scene monitor
Traceback (most recent call last):
  File "/home/lenova/robolab_ws/motion_planning_python_api_tutorial.py", line 188, in <module>
    main()
  File "/home/lenova/robolab_ws/motion_planning_python_api_tutorial.py", line 63, in main
    panda = MoveItPy(node_name="moveit_py_planning_scene")
```

I believe the error is related to the configuration of moveit inside `_robot_moveit_common.launch.py` script. I tried to do it on my own, but unfortunately couldn't understand how can fix it. (sorry I just started with moveit2).

I found a similar issue but related to UR robot  [Testing MoveIt2 Python Bindings with real UR robot #2014 ](https://github.com/moveit/moveit2/issues/2014). 

could you please navigate me and help to run the basic motion_planning_python_api_tutorial.py example for moveit-py

All best

penglongxiang, 2024-06-14:
Hi @adamanov, could you please share more details of the **full error message**, and what **modifications** you have made to [example motion_planning_python_api_tutorial.py](https://github.com/moveit/moveit2_tutorials/blob/main/doc/examples/motion_planning_python_api/scripts/motion_planning_python_api_tutorial.py) for adaptation of xArm?

omukesh, 2025-01-08:
can some one revert on this ??? regarding pybind ?