xArm-Developer/xarm_ros2 #76: Hardware Interface Status Values File Missing

Leopold-Klotz, 2024-06-21:
When trying to build the xarm_controller package, a fatal error occurs with the line: 
#include "hardware_interface/types/hardware_interface_status_values.hpp"

I have gone through the ros2_control package and I don't believe that this file exists. Please let me know how to get around this error, or if I am doing something incorrectly.

penglongxiang, 2024-06-21:
Hi @Leopold-Klotz, have you checked out the correct branch according to your ROS2 version? The default master may not be compatible since ROS2 code changes more between generations.

Leopold-Klotz, 2024-06-21:
Thank you, I apologize I thought that I had tried this but repeating the process fixed my issue.