xArm-Developer/xarm_ros2 #79: Dual setup with gripper fails

mcres, 2024-07-05:
Launching the following dual setup with grippers:

```bash
ros2 launch xarm_moveit_config dual_uf850_moveit_fake.launch.py add_gripper:=true
```

leads to the `ros2_control_node` process being terminated:

```bash
[ros2_control_node-7]   what():  Hardware name XArmGripperSystem is duplicated. Please provide a unique 'name' in the URDF.
```

The fix is as easy as parametrizing each gripper `ros2_control` tag name using the prefix, e.g. [this one](https://github.com/xArm-Developer/xarm_ros2/blob/humble/xarm_description/urdf/gripper/xarm_gripper.ros2_control.xacro#L5).

Of course, this should be extended to the rest of the grippers.

vimior, 2024-07-08:
@mcres 
Thanks for the heads up, we will fix this issue.

vimior, 2024-07-08:
@mcres 
Thanks for pointing this out, we will fix it.