xArm-Developer/xarm_ros2 #94: failed to run the gazebo simulation

bigfacecat553, 2024-09-08:
Hello, I am a new player of your product.

I have already successfully build the project:
```
colcon build
Starting >>> xarm_description
Starting >>> xarm_msgs
Starting >>> xarm_sdk
Starting >>> uf_ros_lib
Starting >>> realsense_gazebo_plugin
Finished <<< xarm_description [0.36s]                                                                                                                                   
Finished <<< realsense_gazebo_plugin [0.45s]                                                                                                              
Finished <<< xarm_sdk [0.49s]                                                                                 
Finished <<< uf_ros_lib [0.79s]                                                     
Finished <<< xarm_msgs [2.98s]                      
Starting >>> xarm_api
Finished <<< xarm_api [0.25s]                      
Starting >>> xarm_controller
Finished <<< xarm_controller [0.18s]                   
Starting >>> xarm_gazebo
Finished <<< xarm_gazebo [0.20s]                   
Starting >>> xarm_moveit_config
Finished <<< xarm_moveit_config [0.13s]                   
Starting >>> xarm_moveit_servo
Starting >>> xarm_planner
Finished <<< xarm_moveit_servo [0.18s]                                                                
Finished <<< xarm_planner [0.30s]                    

Summary: 11 packages finished [4.28s]
```

then when I run the 
```
source install/setup.bash
gazebo -s libgazebo_ros_init.so -s libgazebo_ros_factory.so myworld.world
```

it just report:

```
[spawn_entity.py-2]   warnings.warn(
[spawn_entity.py-2] [INFO] [1725802397.880779174] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-2] [INFO] [1725802397.991853718] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-2] [INFO] [1725802397.992271620] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-2] [ERROR] [1725802428.042439656] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-2] [ERROR] [1725802428.042857081] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-2]: process has died [pid 15440, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -topic robot_description -entity UF_ROBOT -x -0.2 -y -0.5 -z 1.021 -Y 1.571 --ros-args --params-file /tmp/launch_params_kn9kyqg9'].

```

It there something wrong with the path of model? Or anything else?
How can I fix it?

Thanks

penglongxiang, 2024-09-09:
Hi @bigfacecat553, unfortunately we can not reproduce your issue unless provided all the changes made and parameters given, because the original command:
```
gazebo -s libgazebo_ros_init.so -s libgazebo_ros_factory.so myworld.world
```
had nothing to do with `xarm_ros2` models.

bigfacecat553, 2024-09-09:
Sorry, I think I pasted the wrong code slide yesterday. Actually, I didn’t make any changes to the code; I just ran it directly after successfully build：

```
source install/setup.bash
ros2 launch xarm_gazebo lite6_beside_table_gazebo.launch.py
```

and then it just always reports the error that I attach above.

penglongxiang, 2024-09-10:
Hi @bigfacecat553, I have checked with your specified command and everything works normally. Here is my console output:
```shell
jason@jason-B85-HD3:~/dev_ws$ ros2 launch xarm_gazebo lite6_beside_table_gazebo.launch.py
[INFO] [launch]: All log files can be found below /home/jason/.ros/log/2024-09-10-11-10-26-903836-jason-B85-HD3-37986
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [37989]
[INFO] [gzclient-2]: process started with pid [37991]
[INFO] [robot_state_publisher-3]: process started with pid [37993]
[INFO] [spawn_entity.py-4]: process started with pid [37995]
[robot_state_publisher-3] [INFO] [1725937827.947143257] [robot_state_publisher]: got segment link1
[robot_state_publisher-3] [INFO] [1725937827.947879258] [robot_state_publisher]: got segment link2
[robot_state_publisher-3] [INFO] [1725937827.948240662] [robot_state_publisher]: got segment link3
[robot_state_publisher-3] [INFO] [1725937827.948549802] [robot_state_publisher]: got segment link4
[robot_state_publisher-3] [INFO] [1725937827.948851569] [robot_state_publisher]: got segment link5
[robot_state_publisher-3] [INFO] [1725937827.949150790] [robot_state_publisher]: got segment link6
[robot_state_publisher-3] [INFO] [1725937827.949508677] [robot_state_publisher]: got segment link_base
[robot_state_publisher-3] [INFO] [1725937827.951287203] [robot_state_publisher]: got segment link_eef
[robot_state_publisher-3] [INFO] [1725937827.951741827] [robot_state_publisher]: got segment world
[spawn_entity.py-4] [INFO] [1725937828.542791178] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1725937828.543243951] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-4]   warnings.warn(
[spawn_entity.py-4] [INFO] [1725937828.545397155] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1725937828.568233882] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1725937828.568601585] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [INFO] [1725937829.147177313] [spawn_entity]: Calling service /spawn_entity
[spawn_entity.py-4] [INFO] [1725937829.410927646] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [UF_ROBOT]
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 37995]

```
Could you please verify whether your gazebo installation is complete, and why there is output saying `Service %s/spawn_entity unavailable` in your console?

bigfacecat553, 2024-09-10:
Thank you for your patient help. I think it seems a bug of ros. 

https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1442

I rebuilt a workspace and compiled the project, and found that it was executing smoothly.

closed 2024-09-10 by bigfacecat553