xArm-Developer/xarm_ros2 #95: lite6 [Firmware V 2.4] for ros2 humble - Stuttering (Shaking) by using moveit servo with xBox controller

lu4k87, 2024-09-11:
hi after the firmware update from v 2.0.1 to v 2.4 i have a stuttering(shaking) movement when executing moveit servo. 
i use the xbox controller to enter the direction.  
what do i have to do to make moveit servo for ros2 humble (i have successfully installed it as described here). 

i have seen that in git: xArm-Developer/xArm_ros
(not xArm_ros2) the files were updated 2 days ago. 

what is the best way to update the files? (config, urdf, srdf, ...) 
I use:
ros2 launch xarm_moveit_servo lite6_moveit_servo_realmove.launch.py robot_ip:=192.168.1.xxx joystick_type:=1

thanks for help

lu4k87, 2024-09-12:
btw... when iam using: 
>> ros2 launch xarm_moveit_servo lite6_moveit_servo_fake.launch.py joystick_type:=1

I got a: MotionPlanning Status Warn: Requesting initial scene failed.
is there any way to downgrade the firmware to V 2.0.1?

penglongxiang, 2024-09-12:
Hi @lu4k87, after firmware version 2.4.0, the feedback position will be from actual encoders rather than received command, so it may require higher control rate for ros-control, please try editing [this file](https://github.com/xArm-Developer/xarm_ros2/blob/master/xarm_controller/config/lite6_controllers.yaml#L3) and change the parameter `update_rate` to be **125** to **200** (Hz). In this way the control can be smoother. Please have a try and tell us if this works for you.

Regarding to "`MotionPlanning Status Warn: Requesting initial scene failed`" warning in your last post, since this comes from fake/simulation scenario, it has nothing to do with our firmware version. Could you please give us the full output from the terminal in this case?

lu4k87, 2024-09-12:
Thanks for the fast reply, will give u Feedback soon

lu4k87, 2024-09-12:
i changed it, but its still the same problem... maybe i have to chane the "state_publish_rate: 25.0" to 200 Hz too?

**lite6_controllers.yaml**
controller_manager:
  ros__parameters:
    update_rate: 200  # Hz

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    lite6_traj_controller:
      type: joint_trajectory_controller/JointTrajectoryController

lite6_traj_controller:
  ros__parameters:
    command_interfaces:
      - position
      - velocity
    state_interfaces:
      - position
      - velocity
    joints:
      - joint1
      - joint2
      - joint3
      - joint4
      - joint5
      - joint6
    constraints:
      goal_time: 0.5
      stopped_velocity_tolerance: 0.0
      joint1: {trajectory: 1.0, goal: 0.01}
      joint2: {trajectory: 1.0, goal: 0.01}
      joint3: {trajectory: 1.0, goal: 0.01}
      joint4: {trajectory: 1.0, goal: 0.01}
      joint5: {trajectory: 1.0, goal: 0.01}
      joint6: {trajectory: 1.0, goal: 0.01}
    interface_name: position
    state_publish_rate: 25.0
    action_monitor_rate: 10.0

lu4k87, 2024-09-12:
it is necessary to build(colcom build) or/and source my ws after change the update rate?

sorry, iam a beginner

lu4k87, 2024-09-12:
i changed the update_rate to 250 and the movement is much better...
[INFO] [launch]: All log files can be found below /home/labadmin/.ros/log/2024-09-12-17-43-23-806051-labadmin-MacPro-25459
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [25479]
[INFO] [ros2_control_node-2]: process started with pid [25481]
[INFO] [component_container-3]: process started with pid [25483]
[INFO] [spawner-4]: process started with pid [25485]
[INFO] [spawner-5]: process started with pid [25487]
[ros2_control_node-2] [WARN] [1726155807.306227024] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-2] [INFO] [1726155807.306582889] [resource_manager]: Loading hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[ros2_control_node-2] [INFO] [1726155807.319379839] [resource_manager]: Initialize hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[ros2_control_node-2] [INFO] [1726155807.355656811] [UFACTORY.RobotHW]: [192.168.1.175] namespace: /
[ros2_control_node-2] [INFO] [1726155807.355699273] [UFACTORY.RobotHW]: [192.168.1.175] robot_type: lite, hw_ns: ufactory, prefix: , report_type: dev
[ros2_control_node-2] [INFO] [1726155807.356393154] [UFACTORY.RobotHW]: [192.168.1.175] dof: 6, velocity_control: 0, add_gripper: 0, baud_checkset: 1, default_gripper_baud: 2000000
[ros2_control_node-2] [INFO] [1726155807.356715456] [ufactory_driver]: robot_ip=192.168.1.175, report_type=dev, dof=6
[ros2_control_node-2] [INFO] [1726155807.356815238] [ufactory_driver]: baud_checkset: 1, default_gripper_baud: 2000000
[ros2_control_node-2] SDK_VERSION: 1.13.6
[ros2_control_node-2] Tcp control connection successful
[ros2_control_node-2] change protocol identifier to 2
[component_container-3] [INFO] [1726155807.403313454] [xarm_moveit_servo_container]: Load Library: /home/labadmin/ros2_humble/install/robot_state_publisher/lib/librobot_state_publisher_node.so
[component_container-3] [INFO] [1726155807.410412916] [xarm_moveit_servo_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-3] [INFO] [1726155807.410492923] [xarm_moveit_servo_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[component_container-3] [INFO] [1726155807.435045283] [robot_state_publisher]: got segment link1
[component_container-3] [INFO] [1726155807.435122090] [robot_state_publisher]: got segment link2
[component_container-3] [INFO] [1726155807.435139109] [robot_state_publisher]: got segment link3
[component_container-3] [INFO] [1726155807.435151857] [robot_state_publisher]: got segment link4
[component_container-3] [INFO] [1726155807.435164367] [robot_state_publisher]: got segment link5
[component_container-3] [INFO] [1726155807.435176304] [robot_state_publisher]: got segment link6
[component_container-3] [INFO] [1726155807.435188465] [robot_state_publisher]: got segment link_base
[component_container-3] [INFO] [1726155807.435200859] [robot_state_publisher]: got segment link_eef
[component_container-3] [INFO] [1726155807.435213501] [robot_state_publisher]: got segment link_tcp
[component_container-3] [INFO] [1726155807.435225455] [robot_state_publisher]: got segment uflite_vacuum_gripper_link
[component_container-3] [INFO] [1726155807.435238014] [robot_state_publisher]: got segment world
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/robot_state_publisher' in container '/xarm_moveit_servo_container'
[component_container-3] [INFO] [1726155807.444943781] [xarm_moveit_servo_container]: Load Library: /home/labadmin/ros2_humble/install/tf2_ros/lib/libstatic_transform_broadcaster_node.so
[component_container-3] [INFO] [1726155807.445499720] [xarm_moveit_servo_container]: Found class: rclcpp_components::NodeFactoryTemplate<tf2_ros::StaticTransformBroadcasterNode>
[component_container-3] [INFO] [1726155807.445527521] [xarm_moveit_servo_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<tf2_ros::StaticTransformBroadcasterNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/static_tf2_broadcaster' in container '/xarm_moveit_servo_container'
[component_container-3] [INFO] [1726155807.463922031] [xarm_moveit_servo_container]: Load Library: /opt/ros/humble/lib/libservo_node.so
[component_container-3] [INFO] [1726155807.498972187] [xarm_moveit_servo_container]: Found class: rclcpp_components::NodeFactoryTemplate<moveit_servo::ServoNode>
[component_container-3] [INFO] [1726155807.499035897] [xarm_moveit_servo_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<moveit_servo::ServoNode>
[component_container-3] [WARN] [1726155807.509020826] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[component_container-3] extra_arguments=[{'use_intra_process_comms' : True}]
[component_container-3] to the Servo composable node in the launch file
[component_container-3] [INFO] [1726155807.528351834] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0105282 seconds
[component_container-3] [INFO] [1726155807.528411603] [moveit_robot_model.robot_model]: Loading robot model 'UF_ROBOT'...
[... cut ...]
[rviz2-1] [INFO] [1726155812.859461781] [interactive_marker_display_94369123597264]: Service response received for initialization
[component_container-3] [WARN] [1726155822.769881583] [moveit_servo.utilities]: Moving away from a singularity, decelerating
[component_container-3] [WARN] [1726155825.898093956] [moveit_servo.utilities]: Moving closer to a singularity, decelerating
[component_container-3] [WARN] [1726155828.923996514] [moveit_servo.utilities]: Moving closer to a singularity, decelerating

penglongxiang, 2024-09-13:
@lu4k87 you should rebuild the package after parameter changing in ROS2. 

May I confirm the output you posted is from which execution command? Is it from real robot control or fake simulation?

lu4k87, 2024-09-13:
the output comes after the cmd: 
>> ros2 launch xarm_moveit_servo lite6_moveit_servo_realmove.launch.py robot_ip:=192.168.1.xxx joystick_type:=1

i tested with xbox one elite 2 controller. and it works, after i changed the update_rate!

however, if i want to set up the whole setup (ubuntu 22.04+ros2+moveit+xarm_ros2) from scratch as described here in the git, the cmd works: 

>> ros2 launch xarm_moveit_servo lite6_moveit_servo_realmove.launch.py robot_ip:=192.168.1.xxx joystick_type:=1 

and also 

>> $ ros2 launch xarm_moveit_servo lite6_moveit_servo_fake.launch.py joystick_type:=1 

what do i have to do after i have gone through the complete instructions. which files and parameters do i have to change so that i can control both the fake arm and the real arm with the xbox controller on a freshly installed ros2 system?

lu4k87, 2024-09-13:
however, if i want to set up the whole setup (ubuntu 22.04+ros2+moveit+xarm_ros2) from scratch as described here in the git, the cmds: 

>> ros2 launch xarm_moveit_servo lite6_moveit_servo_realmove.launch.py robot_ip:=192.168.1.xxx joystick_type:=1 

and also 

>> $ ros2 launch xarm_moveit_servo lite6_moveit_servo_fake.launch.py joystick_type:=1 dont works

what do i have to do after i have gone through the complete instructions. which files and parameters do i have to change so that i can control both the fake arm and the real arm with the xbox controller on a freshly installed ros2 system?

lu4k87, 2024-09-13:
the is the console log for:
>> ros2 launch xarm_moveit_servo lite6_moveit_servo_fake.launch.py joystick_type:=1

labadmin@MacPro:~$ ros2 launch xarm_moveit_servo lite6_moveit_servo_fake.launch.py joystick_type:=1
[INFO] [launch]: All log files can be found below /home/labadmin/.ros/log/2024-09-13-12-33-59-114189-MacPro-21858
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [21859]
[INFO] [spawner-2]: process started with pid [21861]
[INFO] [ros2_control_node-3]: process started with pid [21863]
[INFO] [spawner-4]: process started with pid [21865]
[ros2_control_node-3] [WARN] [1726223639.502481372] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1726223639.502737967] [resource_manager]: Loading hardware 'uf_robot_hardware/UFRobotFakeSystemHardware' 
[ros2_control_node-3] [INFO] [1726223639.510267751] [resource_manager]: Initialize hardware 'uf_robot_hardware/UFRobotFakeSystemHardware' 
[ros2_control_node-3] [WARN] [1726223639.510305255] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-3] <state_interface name="velocity"> 
[ros2_control_node-3]   <param name="initial_value">0.0</param> 
[ros2_control_node-3] </state_interface>
[ros2_control_node-3] [INFO] [1726223639.510317014] [resource_manager]: Successful initialization of hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [INFO] [1726223639.510364229] [resource_manager]: 'configure' hardware 'uf_robot_hardware/UFRobotFakeSystemHardware' 
[ros2_control_node-3] [INFO] [1726223639.510371931] [resource_manager]: Successful 'configure' of hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [INFO] [1726223639.510378465] [resource_manager]: 'activate' hardware 'uf_robot_hardware/UFRobotFakeSystemHardware' 
[ros2_control_node-3] [INFO] [1726223639.510383611] [resource_manager]: Successful 'activate' of hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [INFO] [1726223639.513774900] [controller_manager]: update rate is 150 Hz
[ros2_control_node-3] [INFO] [1726223639.514617153] [controller_manager]: Successful set up FIFO RT scheduling policy with priority 50.
[spawner-4] [INFO] [1726223639.772828376] [spawner_lite6_traj_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-2] [INFO] [1726223639.791344845] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-3] [INFO] [1726223640.024254236] [controller_manager]: Loading controller 'lite6_traj_controller'
[ros2_control_node-3] [WARN] [1726223640.031727924] [lite6_traj_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[spawner-4] [INFO] [1726223640.035579400] [spawner_lite6_traj_controller]: Loaded lite6_traj_controller
[ros2_control_node-3] [INFO] [1726223640.043245699] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-2] [INFO] [1726223640.048972122] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-3] [INFO] [1726223640.049557413] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-3] [INFO] [1726223640.049629127] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-2] [INFO] [1726223640.068932167] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[rviz2-1] [INFO] [1726223640.110008526] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1726223640.110118528] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[ros2_control_node-3] [INFO] [1726223640.117252898] [controller_manager]: Configuring controller 'lite6_traj_controller'
[ros2_control_node-3] [INFO] [1726223640.117377778] [lite6_traj_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-3] [INFO] [1726223640.117394114] [lite6_traj_controller]: Command interfaces are [position velocity] and state interfaces are [position velocity].
[ros2_control_node-3] [INFO] [1726223640.117409874] [lite6_traj_controller]: Using 'splines' interpolation method.
[ros2_control_node-3] [INFO] [1726223640.117977484] [lite6_traj_controller]: Controller state will be published at 25.00 Hz.
[ros2_control_node-3] [INFO] [1726223640.119080577] [lite6_traj_controller]: Action status changes will be monitored at 10.00 Hz.
[spawner-4] [INFO] [1726223640.142113382] [spawner_lite6_traj_controller]: Configured and activated lite6_traj_controller
[INFO] [spawner-2]: process has finished cleanly [pid 21861]
[rviz2-1] [INFO] [1726223640.220417158] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [spawner-4]: process has finished cleanly [pid 21865]
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one runni [... cut ...]
[rviz2-1] [INFO] [1726223648.621356082] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-1] [INFO] [1726223648.621407446] [moveit_ros_visualization.motion_planning_frame]: group lite6
[rviz2-1] [INFO] [1726223648.621415115] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'lite6' in namespace ''
[rviz2-1] [INFO] [1726223708.626356225] [move_group_interface]: Ready to take commands for planning group lite6.

[image]

Shahar-Nadiv, 2025-06-03:
Hey regarding the previous attached terminal, 
i have an issue with my UF850, which i get a lot of reaching closer to singularity,
80% of positions in the envelop are in singularities according to Moveit, where as in the real robot we manage to move 
in the same direction when using the Ufactory GUI,
why do i reach to singularities that often? any suggestions?