xArm-Developer/xarm_ros2 #96: Using xarm_ros2 without any gazebo dependencies

bchandaka, 2024-09-13:
Hi,

I'm trying to use the ros2 humble packages to control my xarm7 from a jetson orin nx. However, Gazebo(and related packages like gazebo-ros, gazebo-ros2-control, etc.) are not supported on ARM devices like the Jetson. I don't need to use gazebo at all, but I just want to run real robot arm control(Just this command needs to work: `ros2 launch xarm_moveit_config xarm7_moveit_realmove.launch.py robot_ip:=192.168.1.117 add_gripper:=true`).

How can I build the xarm_ros2 packages without any of the gazebo dependencies?

penglongxiang, 2024-09-18:
Hi @bchandaka, you may try removing "`xarm_gazebo`" and "`thirdparty/realsense_gazebo_plugin`" packages, then the rest will not have gazebo dependencies.

gaspatxo, 2024-09-18:
For me this fails in the compilation of `xarm_planner`:

```
Starting >>> xarm_planner
--- stderr: xarm_planner
CMake Error at /home/gas/ws_moveit2/install/moveit_msgs/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package):
  By not providing "Findobject_recognition_msgs.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "object_recognition_msgs", but CMake did not find one.

  Could not find a package configuration file provided by
  "object_recognition_msgs" with any of the following names:

    object_recognition_msgsConfig.cmake
    object_recognition_msgs-config.cmake

  Add the installation prefix of "object_recognition_msgs" to
  CMAKE_PREFIX_PATH or set "object_recognition_msgs_DIR" to a directory
  containing one of the above files.  If "object_recognition_msgs" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  /home/gas/ws_moveit2/install/moveit_msgs/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  /home/gas/ws_moveit2/install/moveit_ros_planning_interface/share/moveit_ros_planning_interface/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/gas/ws_moveit2/install/moveit_ros_planning_interface/share/moveit_ros_planning_interface/cmake/moveit_ros_planning_interfaceConfig.cmake:41 (include)
  CMakeLists.txt:28 (find_package)
```

penglongxiang, 2024-09-18:
@gaspatxo please try installing `ros-$ROS_DISTRO-object-recognition-msgs` package.

gaspatxo, 2024-09-18:
It worked, I did have to install other packages as well:

```
apt install \
ros-humble-octomap-msgs \
ros-humble-srdfdom \
ros-humble-geometric-shapes \
ros-humble-ruckig \
ros-humble-launch-param-builder
```

Note that I am using Moveit2 built from source.

bchandaka, 2024-09-20:
Ok, thanks I got it to work.

closed 2024-09-20 by bchandaka