xArm-Developer/xarm_ros2 #97: Lite 6 - moveit_servo - Ros2 Humble

lu4k87, 2024-09-14:
if i want to set up the whole setup from scratch as described here in the git, the cmds:

>> ros2 launch xarm_moveit_servo lite6_moveit_servo_realmove.launch.py robot_ip:=192.168.1.xxx joystick_type:=1

and also

>> ros2 launch xarm_moveit_servo lite6_moveit_servo_fake.launch.py joystick_type:=1

dont works.
what do i have to do after i have gone through the complete instructions. which files and parameters do i have to change so that i can control both the fake arm and the real robot with the xbox controller on a freshly installed ros2 system? 

Please tell me, which steps i have to do, after a fresh and clear installation of (ros2 humble - Ubuntu (deb packages) + moveit2 + gazebo + gazebo-ros xarm_sdk in "dev_ws/src/xarm_ros2" 
build + source.

Is it necessary to intall the whole "ws_moveit2"? 
if yes, should i put the "ws_moveit2" in the "dev_ws_/src" like "xarm_ros2" or "home/dev_ws" and "home/ws_moveit2"? 

And how to mangage the files from: "ws_moveit2" to "xarm_ros2/moveit_servo" and "xarm_moveit_config" ?

maybe i need to do the moveit setup assistant? but the setup assistant i can only found in "ws_moveit2" ?
thanks for help!

I run: >> ros2 launch xarm_moveit_servo lite6_moveit_servo_fake.launch.py joystick_type:=1

[image]

labadmin@MacPro:$ ros2 launch xarm_moveit_servo lite6_moveit_servo_fake.launch.py joystick_type:=1
[INFO] [launch]: All log files can be found below /home/labadmin/.ros/log/2024-09-13-12-33-59-114189-MacPro-21858
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [21859]
[INFO] [spawner-2]: process started with pid [21861]
[INFO] [ros2_control_node-3]: process started with pid [21863]
[INFO] [spawner-4]: process started with pid [21865]
[ros2_control_node-3] [WARN] [1726223639.502481372] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1726223639.502737967] [resource_manager]: Loading hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [INFO] [1726223639.510267751] [resource_manager]: Initialize hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [WARN] [1726223639.510305255] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example:
[ros2_control_node-3] <state_interface name="velocity">
[ros2_control_node-3] 0.0
[ros2_control_node-3] </state_interface>
[ros2_control_node-3] [INFO] [1726223639.510317014] [resource_manager]: Successful initialization of hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [INFO] [1726223639.510364229] [resource_manager]: 'configure' hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [INFO] [1726223639.510371931] [resource_manager]: Successful 'configure' of hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [INFO] [1726223639.510378465] [resource_manager]: 'activate' hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [INFO] [1726223639.510383611] [resource_manager]: Successful 'activate' of hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'
[ros2_control_node-3] [INFO] [1726223639.513774900] [controller_manager]: update rate is 150 Hz
[ros2_control_node-3] [INFO] [1726223639.514617153] [controller_manager]: Successful set up FIFO RT scheduling policy with priority 50.
[spawner-4] [INFO] [1726223639.772828376] [spawner_lite6_traj_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-2] [INFO] [1726223639.791344845] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-3] [INFO] [1726223640.024254236] [controller_manager]: Loading controller 'lite6_traj_controller'
[ros2_control_node-3] [WARN] [1726223640.031727924] [lite6_traj_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[spawner-4] [INFO] [1726223640.035579400] [spawner_lite6_traj_controller]: Loaded lite6_traj_controller
[ros2_control_node-3] [INFO] [1726223640.043245699] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-2] [INFO] [1726223640.048972122] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-3] [INFO] [1726223640.049557413] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-3] [INFO] [1726223640.049629127] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-2] [INFO] [1726223640.068932167] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[rviz2-1] [INFO] [1726223640.110008526] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1726223640.110118528] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[ros2_control_node-3] [INFO] [1726223640.117252898] [controller_manager]: Configuring controller 'lite6_traj_controller'
[ros2_control_node-3] [INFO] [1726223640.117377778] [lite6_traj_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-3] [INFO] [1726223640.117394114] [lite6_traj_controller]: Command interfaces are [position velocity] and state interfaces are [position velocity].
[ros2_control_node-3] [INFO] [1726223640.117409874] [lite6_traj_controller]: Using 'splines' interpolation method.
[ros2_control_node-3] [INFO] [1726223640.117977484] [lite6_traj_controller]: Controller state will be published at 25.00 Hz.
[ros2_control_node-3] [INFO] [1726223640.119080577] [lite6_traj_controller]: Action status changes will be monitored at 10.00 Hz.
[spawner-4] [INFO] [1726223640.142113382] [spawner_lite6_traj_controller]: Configured and activated lite6_traj_controller
[INFO] [spawner-2]: process has finished cleanly [pid 21861]
[rviz2-1] [INFO] [1726223640.220417158] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [spawner-4]: process has finished cleanly [pid 21865]
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[... 38 lines cut ...]
[component_container-5] [INFO] [1726223641.736408943] [joy_node]: Opened joystick: Xbox One S Controller. deadzone: 0.050000
[rviz2-1] [ERROR] [1726223643.346331270] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [INFO] [1726223643.354769829] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-1] [ERROR] [1726223643.503219304] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-1]
[rviz2-1] [INFO] [1726223643.508627503] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00335508 seconds
[rviz2-1] [INFO] [1726223643.508663943] [moveit_robot_model.robot_model]: Loading robot model 'UF_ROBOT'...
[rviz2-1] [INFO] [1726223643.508676630] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-1] [INFO] [1726223643.608916920] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-1] [INFO] [1726223643.609376546] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-1] [ERROR] [1726223643.610306720] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-1]
[rviz2-1] [ERROR] [1726223643.611978853] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-1]
[... cut ...]
[rviz2-1] [INFO] [1726223648.621356082] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-1] [INFO] [1726223648.621407446] [moveit_ros_visualization.motion_planning_frame]: group lite6
[rviz2-1] [INFO] [1726223648.621415115] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'lite6' in namespace ''
[rviz2-1] [INFO] [1726223708.626356225] [move_group_interface]: Ready to take commands for planning group lite6.

penglongxiang, 2024-09-14:
Hi @lu4k87 we will be working on this issue and hopefully will fix it next week.

lu4k87, 2024-09-14:
hey, thanks for the reply.

it would be very nice if it would work somehow... i need the xbox control for my thesis at the end of the month :/
(simulation in Rviz + realrobot movement)

the following steps work so far:
3.1 Install ROS2
   -> for Humble
3.2 Install Moveit2
3.3 Install Gazebo
3.4 Install gazebo_ros_pkgs
4 How To Use
4.1 Create a workspace
4.2 Obtain source code of “xarm_ros2” repository
4.3 Update “xarm_ros2” repository
4.4 Install dependencies
4.5 Build xarm_ros2

if you get this fixed, can you please include a short detailed step by step guide where every single step is listed? for a freshly set up ros2 humble system.

many thanks and good luck, hope as soon as possible

penglongxiang, 2024-09-18:
Hi @lu4k87 please update the latest code and try again. As for your doc request, we do have other development tasks going on recently, and we will optimize our doc for more details when we got time for it, thanks for your understanding.

lu4k87, 2024-09-18:
Hey penglongxiang
thank you very much for the fast processing. the control with the xbox controller for the real arm + simulation works again!