xArm-Developer/xarm_ros2 #98: How to control effort on joints?

lnss1-ua, 2024-09-19:
Hello, I'm new to ros2 and I'm having a tough time trying to send/pub trajectories specifying effort. I always get the same message:
[lite6_traj_controller]: Trajectories with effort fields are currently not supported.

For context I'm using ubuntu 22.04 and ros2 humble obv. I'm working with the lite6.

I don't know if i had to load some controller or another before or if it is something else. I checked that there's a controller called effort_controllers/JointGroupEffortController but I can't (probably don't know) how to activate it. 

Thanks in advanced.

penglongxiang, 2024-09-20:
Hi @lnss1-ua, we have no joint effort/torque control interface for our real robot. So we do not support this type of controller in ROS.

lnss1-ua, 2024-09-20:
So there's no way I can send an exact effort to each joint?

penglongxiang, 2024-09-20:
> So there's no way I can send an exact effort to each joint?

Right, we only have position and velocity interface. There is no torque sensor on the joints.

lnss1-ua, 2024-09-20:
By sensor you mean sensors which i could send effort right? Because obv I can get the effort with the /joints_state topic. (sorry for asking this much just wanted to learn :D)

penglongxiang, 2024-09-20:
In order to implement a torque control loop, there has to be a torque sensor on the joint to feedback the **accurate** torque. We do not have this in our design.

The effort info from `/joint_states` topic is actually **estimated value** calculated from motor current sensor data and robot dynamic model, it may not be very accurate since there is friction in the middle which is hard to model accurately, the dynamic model would also have some error more or less. So effort from `/joint_states` is just for reference or some usage that does not require high accuracy, but not directly for joint torque control.

lnss1-ua, 2024-09-20:
Okey Thank you very much :D have a nice day

closed 2024-09-20 by lnss1-ua