xArm-Developer/xarm_ros2 #105: What's joint model group name of uf850

ogios, 2024-10-17:
Sorry i'm new to this.
I'm trying to convert moveit2 tutorial `panda_arm` to `uf850`, so basically it's this:
```python
self.panda = MoveItPy(node_name="moveit_uf850", config_dict=moveit_config)
self.panda_arm = self.panda.get_planning_component("uf850") # The error occurs here
# self.panda.get_planning_component()
self.planning_scene_monitor: PlanningSceneMonitor = (
    self.panda.get_planning_scene_monitor()
)
```
The error goes with:
```
self.panda_arm = self.panda.get_planning_component("uf850")
RuntimeError: Could not find joint model group 'uf850'.
```
I saw that in `xarm_moveit_config/srdf/_uf850_macro.srdf.xacro` it says group name is `{prefix}uf850` and the prefix is actually an empty string so the group name should be `uf850`? But it do not work, i've tried also `xarm/controllers_initial_group_+xarm&uf850...`, none of them works.

So how can i solve it?

penglongxiang, 2024-10-17:
Hi @ogios, it seems '`uf850`' is correct for the group name. Could you please show us **the specific procedures to reproduce the issue**? Like what is the launch file and arguments given before executing this python script? It can imply what nodes and parameters are loaded and running in the background. And could you show us the full python code (can be a minimum version just to catch the error)?

ogios, 2024-10-17:
@penglongxiang Thanks! 
And as i go through the uf_ros_lib MoveItConfigBuilder, i found that it's because i should pass in the `robot_type` & `dof` as xarco arguments.
Now i can get this work.

closed 2024-10-17 by ogios