xArm-Developer/xarm_ros2 #106: xarm_api c++ example ros2

Murali25111993, 2024-10-24:
Hello,

 I am trying to write equivalent ros2 c++ code for the [findobj_grasp_xarm_api.cpp](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_vision/d435i_xarm_setup/src/findobj_grasp_xarm_api.cpp).
 
 I have attached the code here. 
[findobj_grasp_xarm_api.txt](https://github.com/user-attachments/files/17501794/findobj_grasp_xarm_api.txt)

However , when I try to run the code, :

1. ros2 launch xarm_api xarm6_driver.launch.py robot_ip:= 192.168.1.233

2. ros2 run xarm_vision xarm_simple_planner

my robot is not moving. 

[image]

Kindly check the code and provide solution.

Thanks

Murali25111993, 2024-10-29:
I have managed to solve the issue. However I cannot able to set the vacuum gripper

terminal 1:
ros2 launch xarm_api xarm6_driver.launch.py robot_ip:= 192.168.1.233 [add_vacuum_gripper:=true]

I could not able to find ros2 service list and ros2 node list for vacuum gripper

I have attached screenshot

[image]
[image]

penglongxiang, 2024-10-29:
Please follow the instruction [here](https://github.com/xArm-Developer/xarm_ros2/tree/humble?tab=readme-ov-file#54-xarm_api) of "service" part. Create your `xarm_api/config/xarm_user_params.yaml` file (in case xarm_user_params.yaml is overwritten in future repo update) and enable the interested services, in your case it would be setting "`set_vacuum_gripper`" to be `true`. This mechanism is different from our ros1 repository.

Murali25111993, 2024-10-29:
Thanks, This sorted my problem. 

This issue can be closed now.

closed 2024-10-29 by penglongxiang