xArm-Developer/xarm_ros2 #107: St12system_error occurred while initializing hardware 'uf_robot_hardware/UFRobotFakeSystemHardware'

DaniGarciaLopez, 2024-10-28:
When I launch:
```
ros2 launch xarm_moveit_config xarm6_moveit_fake.launch.py
```
I get:
```
[ros2_control_node-6] [INFO] [1730114168.639637289] [controller_manager.resource_manager]: Initialize hardware 'uf_robot_hardware/UFRobotFakeSystemHardware' 
[ros2_control_node-6] [ERROR] [1730114168.639695269] [controller_manager.resource_manager]: Exception of type : St12system_error occurred while initializing hardware 'uf_robot_hardware/UFRobotFakeSystemHardware': Invalid argument
[ros2_control_node-6] [WARN] [1730114168.639739698] [controller_manager.resource_manager]: System hardware component 'uf_robot_hardware/UFRobotFakeSystemHardware' from plugin 'uf_robot_hardware/UFRobotFakeSystemHardware' failed to initialize.
```

After this, `ros2_control_node` crashes and the robot doesn't follow the trajectory using the fake hardware. I'm using ros2 jazzy with the latest binaries from ros2_control and rolling branch from this repo. Any idea how to fix this?

vimior, 2024-11-04:
@DaniGarciaLopez 
We are unable to reproduce this issue for now. You can try updating the system and getting the latest code to recompile. We are also trying to reproduce this issue.

DaniGarciaLopez, 2024-11-04:
I updated all the packages, and it's working now! Some packages might have been updated during this time, which resolved the issue. Thanks for looking into it @vimior.

closed 2024-11-04 by DaniGarciaLopez