xArm-Developer/xarm_ros2 #109: Joint limits in URDF do not match xARM 7 hardware limits.

J-Burgess, 2024-12-20:
Hi, I hope you are well. 

We are using the latest version of the xarm_ros2 repository in the Humble tree. 

I am running into some issues with planning new trajectories which involve the robot stretching to another bench behind it. If I control the robot manually or in Ufactory Studio the joints are well within the limits and the motion is possible. However, with ros2 moveit  I get out of bound and joint position errors frequently. 

ros2 launch xarm_moveit_config xarm7_moveit_realmove.launch.py robot_ip:=192.168.1.209 add_bio_gripper:=true

And then when requesting a new motion plan I get the following failures:

[move_group-4] [WARN] [1734668167.535488317] [moveit_ros.fix_start_state_bounds]: Joint 'joint4' from the starting state is outside bounds by a significant margin: [3.67145 ] should be in the range [-0.11 ], [3.11018 ] but the error above the ~start_state_max_bounds_error parameter (currently set to 0.100000)

Furthermore, if I start from within boundaries the robot then constrains its motion and successfully executes all trajectories but within the limits of [-0.11] and [3.11018] for joint 4 which does not reflect the real trajectory desired. I get warnings during these executions such as:

[move_group-4] [WARN] [1734655345.289019343] [moveit_kinematic_constraints.kinematic_constraints]: Joint joint4 is constrained to be above the maximum bounds. Assuming maximum bounds instead. 

Looking in the URDF files within the /src directory after colcon build I can see:
* dev_ws/src/xarm_ros2/xarm_description/urdf/xarm7/xarm7.ros2_control.xacro

Which contains the tag:
```
  <xacro:macro name="xarm7_ros2_control" params="prefix
    velocity_control:='false'
    ros2_control_plugin:='uf_robot_hardware/UFRobotSystemHardware'
    hw_ns:='xarm' add_gripper:='false' add_bio_gripper:='false'
    robot_ip:='' report_type:='normal' baud_checkset:='true' default_gripper_baud:=2000000
    joint1_lower_limit:=${-2.0*pi}  joint1_upper_limit:=${2.0*pi}
    joint2_lower_limit:=${-2.059}  joint2_upper_limit:=${2.0944}
    joint3_lower_limit:=${-2.0*pi}  joint3_upper_limit:=${2.0*pi}
    joint4_lower_limit:=${-0.19198}  joint4_upper_limit:=${3.927}
    joint5_lower_limit:=${-2.0*pi}  joint5_upper_limit:=${2.0*pi}
    joint6_lower_limit:=${-1.69297}  joint6_upper_limit:=${pi}
    joint7_lower_limit:=${-2.0*pi}  joint7_upper_limit:=${2.0*pi}">
    <ros2_control name="${prefix}${ros2_control_plugin}" type="system">
```

Where it appears the limits of joint 4 should actually be -0.19 and 3.927 and not -0.11 and 3.11018?

However, there is another URDF file:
*  dev_ws/src/xarm_ros2/xarm_description/urdf/xarm7/xarm7_robot_macro.xacro

`    
    <xacro:if value="${is_ros2}">
      <xacro:include filename="$(find xarm_description)/urdf/xarm7/xarm7.ros2_control.xacro" />
    </xacro:if>
    <xacro:include filename="$(find xarm_description)/urdf/xarm7/xarm7.urdf.xacro" />
    <xacro:include filename="$(find xarm_description)/urdf/xarm7/xarm7.transmission.xacro" />
    <xacro:include filename="$(find xarm_description)/urdf/xarm7/xarm7.gazebo.xacro" />

    <xacro:if value="${limited}">
      <xacro:if value="${is_ros2}">
        <xacro:xarm7_ros2_control prefix="${prefix}"
          velocity_control="${velocity_control}"
          ros2_control_plugin="${ros2_control_plugin}"
          hw_ns="${hw_ns}" add_gripper="${add_gripper}" add_bio_gripper="${add_bio_gripper}"
          robot_ip="${robot_ip}" report_type="${report_type}"
          baud_checkset="${baud_checkset}" default_gripper_baud="${default_gripper_baud}"
          joint1_lower_limit="${-pi*0.99}" joint1_upper_limit="${pi*0.99}"
          joint2_lower_limit="${-2.18}" joint2_upper_limit="${2.18}"
          joint3_lower_limit="${-pi*0.99}" joint3_upper_limit="${pi*0.99}"
          joint4_lower_limit="${-0.11}" joint4_upper_limit="${pi*0.99}"
          joint5_lower_limit="${-pi*0.99}" joint5_upper_limit="${pi*0.99}"
          joint6_lower_limit="${-1.75}" joint6_upper_limit="${pi*0.99}"
          joint7_lower_limit="${-pi*0.99}" joint7_upper_limit="${pi*0.99}"/>
      </xacro:if>
      <xacro:xarm7_urdf prefix="${prefix}"
        joint1_lower_limit="${-pi*0.99}" joint1_upper_limit="${pi*0.99}"
        joint2_lower_limit="${-2.18}" joint2_upper_limit="${2.18}"
        joint3_lower_limit="${-pi*0.99}" joint3_upper_limit="${pi*0.99}"
        joint4_lower_limit="${-0.11}" joint4_upper_limit="${pi*0.99}"
        joint5_lower_limit="${-pi*0.99}" joint5_upper_limit="${pi*0.99}"
        joint6_lower_limit="${-1.75}" joint6_upper_limit="${pi*0.99}"
        joint7_lower_limit="${-pi*0.99}" joint7_upper_limit="${pi*0.99}"
        model_num="${model_num}"
        inertial_params_filename="${inertial_params_filename}"
        kinematics_params_filename="${kinematics_params_filename}"
        />`
        
    Which does seem to set joint 4 limits as pi*0.99 which is ~0.311? I have tried modifying these lines in the urdf and restarting my ros2 but I always get stuck with the joint constraint errors between [-0.11] and [3.11]. We need to be able to use the whole motion of the xARM 7 for our protocols. Could you help me set the limits required for this? Perhaps I am editing the wrong files or I need to make the changes elsewhere and then rebuild /src? 

Thanks in advance!

James.

penglongxiang, 2024-12-23:
Hi @J-Burgess, Sorry we did not specify in ros2 doc for this behavior. The max clamp limit of `pi*0.99` of the joints will take effect by parameter `limited`, and it defaults to be **true** in upper calling launch files for our moveit applications. It was due to the unexpected solutions (not the shortest solution, takes detours) of default IK solver used by Moveit when joint has a limit greater than 180 degrees (or `pi` in radians). This behavior was observed in our ROS1 development several years ago and we are not sure if this issue has been solved by KDL yet. We can do some more test and set the default to false if this issue no more exists.

Currently if you wish to use full joint range in Moveit, please specify `limited:=false` in the launch command line. Sorry for the inconvenience.

J-Burgess, 2024-12-23:
Hey @penglongxiang thank you for the rapid response and no problem! I actually had some success with modifying the pi*0.99 to simply ${3.297} and then restarting all terminal sessions and re-connecting the robot. The motions are now successful within the new boundaries, I tested up to [3.8] on joint 4. 

Thanks for the information on limited:=false. I will keep that in mind if any future issues occur. 

Cheers,
James

closed 2024-12-23 by J-Burgess