xArm-Developer/xarm_ros2 #111: minic_joint_plugin doesn't support Ign gazebo version

Chen-yanyi, 2025-01-03:
[ruby $(which ign) gazebo-2] Library [/home/yeahychen/cobot_ws/install/roboticsgroup_upatras_gazebo_plugins/lib/libroboticsgroup_upatras_gazebo_mimic_joint_plugin.so] does not export any plugins. The symbol [IgnitionPluginHook] is missing, or it is not externally visible.
[ruby $(which ign) gazebo-2] Library [/home/yeahychen/cobot_ws/install/roboticsgroup_upatras_gazebo_plugins/lib/libroboticsgroup_upatras_gazebo_mimic_joint_plugin.so] does not export any plugins. The symbol [IgnitionPluginHook] is missing, or it is not externally visible.
[ruby $(which ign) gazebo-2] Library [/home/yeahychen/cobot_ws/install/roboticsgroup_upatras_gazebo_plugins/lib/libroboticsgroup_upatras_gazebo_mimic_joint_plugin.so] does not export any plugins. The symbol [IgnitionPluginHook] is missing, or it is not externally visible.
[ruby $(which ign) gazebo-2] Library [/home/yeahychen/cobot_ws/install/roboticsgroup_upatras_gazebo_plugins/lib/libroboticsgroup_upatras_gazebo_mimic_joint_plugin.so] does not export any plugins. The symbol [IgnitionPluginHook] is missing, or it is not externally visible.
[ruby $(which ign) gazebo-2] Library [/home/yeahychen/cobot_ws/install/roboticsgroup_upatras_gazebo_plugins/lib/libroboticsgroup_upatras_gazebo_mimic_joint_plugin.so] does not export any plugins. The symbol [IgnitionPluginHook] is missing, or it is not externally visible.
[image]
The `minic_joint_plugin` appears to be incompatible with the Ignition Gazebo version. Although I attempted to rewrite the plugin content to ensure compatibility with Ignition, it unfortunately didn't produce any results. I also observed that your launch and XACRO files accommodate options for the Ignition version. However, I couldn’t locate a corresponding `_robot_beside_table_ignition.launch.py` file within the Gazebo launch directory. Have you considered addressing these two points in recent time?
[image]

    # robot gazebo launch
    # xarm_gazebo/launch/_robot_beside_table_ignition.launch.py
    robot_gazebo_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(PathJoinSubstitution([FindPackageShare('xarm_gazebo'), 'launch', '_robot_beside_table_ignition.launch.py'])),
        launch_arguments={
            'dof': dof,
            'robot_type': robot_type,
            'prefix': prefix,
            'moveit_config_dump': moveit_config_dump,
            'load_controller': 'true',
            'show_rviz': 'true',
            'no_gui_ctrl': no_gui_ctrl,
        }.items(),
    )

vimior, 2025-01-07:
@Chen-yanyi 
The jazzy version is not fully compatible yet, and the gazebo plugin is not available yet. You may consider using the humble version.
In addition, it is not recommended to use the startup file starting with _ line directly, as the content inside may not be fully supported. The latest jazzy version no longer supports "Gazebo Classic" but supports "Gazebo Harmonic". 
Regarding the startup of ignition, here is a _robot_beside_table_ignition.launch.py ​​for reference.
[_robot_beside_table_ignition.launch.zip](https://github.com/user-attachments/files/18330855/_robot_beside_table_ignition.launch.zip)

Chen-yanyi, 2025-01-10:
> @Chen-yanyi The jazzy version is not fully compatible yet, and the gazebo plugin is not available yet. You may consider using the humble version. In addition, it is not recommended to use the [...]

hi Dwen, thanks for your reply. My gazebo version is Ignition Fortress, and I'm loading xarm into my cobot, installing it on our mobile robot. I've already completed the setup launch files for the ignition version, and everything is fine. The main problem is controlling the xarm gripper in gazebo. I'm using the mimic joint plugin, but it doesn't support the ign gazebo version. I tried modifying it myself, but it wasn't successful. Do you have any good suggestions?
[image]

Chen-yanyi, 2025-01-11:
[image]

Through referring to the robotiq gripper documentation, I have solved the problem. Thank you very much for your previous help.