xArm-Developer/xarm_ros2 #112: Running XArm6 at high speeds

JohannesPankert, 2025-01-08:
Hello

I am controlling an XArm6 through ros2_control on ROS2 Jazzy in joint servo mode. It works fine at lower speeds but when I try to get close to the maximum speed of 180 deg per seconds, the arm throws a speed exceeds limit error:

```
[ros2_control_node-6] [WARN] [1736343614.056050201] [UFACTORY.RobotHW]: [192.168.1.197] set_servo_angle_j, ret= 1
[ros2_control_node-6] [WARN] [1736343614.075464076] [UFACTORY.RobotHW]: [192.168.1.197] Robot State detected! State: 5
[ros2_control_node-6] [ERROR] [1736343614.137464865] [UFACTORY.RobotHW]: [192.168.1.197] UFACTORY Error detected! Code C24 -> [ Speed Exceeds Limit ] 
```
It does not always happen at the same speed but I usually around 140 deg/s or 2.4 rad/s.

Is there a way to operate the arm at 180 deg/s?

[image]

JohannesPankert, 2025-01-08:
sorry, I posted prematurely. Running the robot in velocity mode solves it.

penglongxiang, 2025-01-09:
Hi, actually for safety consideration, there will be a **linear speed limit under servo mode (1)**. The default value is **1.2 x** Max_linear_velocity (1000 mm/s for xarm). It may be triggered when the arm is stretched out and moving at a high joint speed.

We have an API to tune this value in our Python SDK, please refer [here](https://github.com/xArm-Developer/xArm-Python-SDK/blob/9f6ac54c121324ea1c8808b7884fc076211fab5a/xarm/wrapper/xarm_api.py#L3994), however it has not been ported to ROS2 service yet. Please use this python method to change and save this configuration if you still need to operate at higher speed in Mode 1.