xArm-Developer/xarm_ros2 #115: [jazzy] Update to ROS2 control package broke xarm_controller

Sainna, 2025-02-05:
When using **ROS2 Jazzy**, commit 283f6c6 in [ROS2 control](https://github.com/ros-controls/ros2_control/) removed the [visibility_control.h](https://github.com/ros-controls/ros2_control/commit/283f6c67dfa31564646da17d2cb746b929f9a6af#diff-2e5d037d3aae7790f13e6e37fb5439c7b567997f1faf65261d754f917f10ad14).

This led to build errors like this in my case:

> --- stderr: xarm_controller
> In file included from /root/dev_ws/src/xarm_ros2/xarm_controller/src/hardware/uf_robot_system_hardware.cpp:9:
> /root/dev_ws/src/xarm_ros2/xarm_controller/include/xarm_controller/hardware/uf_robot_system_hardware.h:24:10: fatal error: hardware_interface/visibility_control.h: No such file or directory
>   24 | #include "hardware_interface/visibility_control.h"
>        |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>  compilation terminated.

Simply removing the `#include "hardware_interface/visibility_control.h"` in [xarm_controller/include/xarm_controller/hardware/uf_robot_system_hardware.h](https://github.com/xArm-Developer/xarm_ros2/blob/jazzy/xarm_controller/include/xarm_controller/hardware/uf_robot_system_hardware.h) was enough to solve the issue for me. 

I'm not sure of the full implications but, letting you know just in case.

JackLeerson, 2025-02-06:
Sam issue here

Sainna, 2025-02-12:
[797c8a4](https://github.com/xArm-Developer/xarm_ros2/commit/797c8a453ff0e34527abd582009c162e60d17f1f) seems to have fixed the issue. Closing.

closed 2025-02-12 by Sainna